NVIDIA/Isaac-GR00T #1: How to deploy it on a real robot?

becauseofAI, 2025-03-19:


youliangtan, 2025-03-19:
Follow this tutorial: https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/5_policy_deployment.md. and [this](https://github.com/NVIDIA/Isaac-GR00T/tree/main?tab=readme-ov-file#4-evaluation) client-server example: 

We provide an example script for [so100](https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/examples/eval_gr00t_so100.py) that treats robot as client and policy as a service.

BouajilaHamza, 2025-03-20:
I do not understant,
Can the robot move directly by just running the model ? 
can it be deployed without ROS ?

youliangtan, 2025-03-20:
@BouajilaHamza yes you can, you can directly load the model and use the `action_chunk = Gr00tPolicy.get_action(obs)` api directly. It is mentioned on the main readme. 

ROS is not used.

BouajilaHamza, 2025-03-20:
@youliangtan okeey thank you

closed 2025-04-18 by youliangtan