NVIDIA/Isaac-GR00T #8: Deploy for Aloha 2 robot arm

mclau154, 2025-03-19:
Is there a way to deploy for Aloha 2 robot arm? Google recently showed us Gemini with Aloha 2 with their Mujoco system but it is not open source so can we similarly apply Isaac-GR00T to control and learn with this robot arm?

HuFY-dev, 2025-03-20:
We do not have existing code to deploy the policy on Aloha 2. You can follow the [SO-100 example](https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/examples/eval_gr00t_so100.py) and try to write an `AlohaRobotClient` to run with our policy.

Relevant issue: #1

HuFY-dev, 2025-03-21:
Hi @mclau154, let me know if you have any further questions. If everything's resolved, feel free to close the issue.

closed 2025-03-25 by mclau154