NVIDIA/Isaac-GR00T #24: Minimum GPU Requirement

Huisouan, 2025-03-20:
What is the minimum GPU requirement for adjusting/fine-tuning the G1 model？

im-yousuf, 2025-03-20:
The minimum GPU requirement for fine-tuning a model depends on its size and the fine-tuning strategy.

yumeminami, 2025-03-20:
+1

i tried to run the tutor script: 
```bash
python scripts/gr00t_finetune.py \
   --dataset-path ./getting_started/demo_data/so100_strawberry_grape/ \
   --num-gpus 1 \
   --output-dir ~/so100-checkpoints  \
   --max-steps 2000 \
   --data-config so100 \
   --video-backend torchvision_av
```

but the result is `torch.OutOfMemoryError`

`nvidia-smi` output:

```bash
Thu Mar 20 16:23:21 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.120                Driver Version: 550.120        CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090 D      Off |   00000000:01:00.0 Off |                  Off |
| 30%   45C    P8             24W /  425W |    1558MiB /  24564MiB |      1%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090 D      Off |   00000000:05:00.0 Off |                  Off |
| 30%   49C    P2            156W /  425W |   16339MiB /  24564MiB |     76%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2770      G   /usr/lib/xorg/Xorg                            488MiB |
|    0   N/A  N/A      2909      G   /usr/bin/gnome-shell                           24MiB |
|    0   N/A  N/A      6272      G   ...seed-version=20250124-140855.299000         50MiB |
|    0   N/A  N/A    194063      G   ...ures=SpareRendererForSitePerProcess         17MiB |
|    0   N/A  N/A   1102365      C   python                                        348MiB |
|    0   N/A  N/A   1388749      G   ...ions/Zotero_linux-x86_64/zotero-bin         14MiB |
|    0   N/A  N/A   1399214      C   /usr/NX/bin/nxnode.bin                        383MiB |
|    0   N/A  N/A   2022893      G   ...erProcess --variations-seed-version         51MiB |
|    0   N/A  N/A   2963172      G   ...erProcess --variations-seed-version         35MiB |
|    0   N/A  N/A   3241658      G   .../noetic/lib/plotjuggler/plotjuggler         63MiB |
|    1   N/A  N/A      2770      G   /usr/lib/xorg/Xorg                              4MiB |
|    1   N/A  N/A   1102365    C+G   python                                      16306MiB |
+-----------------------------------------------------------------------------------------+
```

please inform the minimum GPU requirement.

youliangtan, 2025-03-20:
You can decrease the `batch-size` if OOM.

If you are not tuning the visual module, with `--no-tune-visual`, the `batch-size` can be signifantly increased

Example:
```bash
python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 2 --no-tune-visual --batch-size 8
```

HuBocheng, 2025-03-21:
Firstly, thanks for the amazing work on GR00T!

I also encountered **out-of-memory (OOM) errors** when trying to run the fine-tuning on the demo dataset with **8×3090 GPUs.**

To reduce memory usage, I’ve already made the following changes:
	•	Set `batch_size` = 1
	•	Set `tune_visual` = False

Despite this, I still encounter OOM errors during training. Also, according to my calculation, the number of trainable parameters is only about 0.8B, which seems manageable.

```
GPU memory before training: 8.159651279449463 GB
Fine-tuning model paramenters in total: 857884160
TensorBoard logs will be saved to: tmp/gr00t/runs
max_steps is given, it will override any value given in num_train_epochs
train dataloader length: 262
train dataset length: 2096
```

Here is my current Config class for reference:

```python
class Config:
    """Configuration for GR00T model fine-tuning."""
# Dataset parameters
dataset_path: str = "./demo_data/robot_sim.PickNPlace"
output_dir: str = "./tmp/gr00t"
data_config: str = "gr1_arms_only"

# Training parameters
batch_size: int = 1
max_steps: int = 10000
num_gpus: int = 8
save_steps: int = 500

# Model parameters
base_model_path: str = "./checkpoints/GR00T-N1-2B"
tune_llm: bool = False
tune_visual: bool = False
tune_projector: bool = True
tune_diffusion_model: bool = True
resume: bool = False

# Advanced training parameters
learning_rate: float = 1e-4
weight_decay: float = 1e-5
warmup_ratio: float = 0.05
dataloader_num_workers: int = 8
report_to: str = "wandb"

# Data loading parameters
embodiment_tag: str = "new_embodiment"
video_backend: str = "decord"
```

Has anyone successfully run the fine-tuning code with a GPU setup similar in total memory to 1×H100?

Any suggestions or insights would be greatly appreciated.

Huisouan, 2025-03-21:
May I ask if the example code can be performed with DeepSpeed to save some GPU memory? According to my current test results, it seems that fine-tuning can be carried out with a 4090 24G without the vision backbone. However, not fine-tuning the vision part may lead to suboptimal fine-tuning results.

zjzhang1999, 2025-03-24:
> Firstly, thanks for the amazing work on GR00T!
> 
> I also encountered **out-of-memory (OOM) errors** when trying to run the fine-tuning on the demo dataset with **8×3090 GPUs.**
> 
> To reduce memory usage, I’ve already made the following changes: • Set `batch_size` = 1 • Set `tune_visual` = False
> 
> Despite this, I still encounter OOM errors during training. Also, according to my calculation, the number of trainable parameters is only about 0.8B, which seems manageable.
> 
> ```
> GPU memory before training: 8.159651279449463 GB
> Fine-tuning model paramenters in total: 857884160
> TensorBoard logs will be saved to: tmp/gr00t/runs
> max_steps is given, it will override any value given in num_train_epochs
> train dataloader length: 262
> train dataset length: 2096
> ```
> 
> Here is my current Config class for reference:
> 
> class Config:
>     """Configuration for GR00T model fine-tuning."""
> # Dataset parameters
> dataset_path: str = "./demo_data/robot_sim.PickNPlace"
> output_dir: str = "./tmp/gr00t"
> data_config: str = "gr1_arms_only"
> 
> # Training parameters
> batch_size: int = 1
> max_steps: int = 10000
> num_gpus: int = 8
> save_steps: int = 500
> 
> # Model parameters
> base_model_path: str = "./checkpoints/GR00T-N1-2B"
> tune_llm: bool = False
> tune_visual: bool = False
> tune_projector: bool = True
> tune_diffusion_model: bool = True
> resume: bool = False
> 
> # Advanced training parameters
> learning_rate: float = 1e-4
> weight_decay: float = 1e-5
> warmup_ratio: float = 0.05
> dataloader_num_workers: int = 8
> report_to: str = "wandb"
> 
> # Data loading parameters
> embodiment_tag: str = "new_embodiment"
> video_backend: str = "decord"
> 
> Has anyone successfully run the fine-tuning code with a GPU setup similar in total memory to 1×H100?
> 
> Any suggestions or insights would be greatly appreciated.

Same problems,find any practical solutions?

zjzhang1999, 2025-03-24:
> You can decrease the `batch-size` if OOM.
> 
> If you are not tuning the visual module, with `--no-tune-visual`, the `batch-size` can be signifantly increased
> 
> Example:
> 
> python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 2 --no-tune-visual --batch-size 8

Thx for the demo intro. may I know why after I set --no-tune-visual --batch-size 1 when fine-tune the dataset still encounter grad is nan error like the screenshots showed. 

[image]

navono, 2025-03-27:
4 4090D ,
> python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 4 --no-tune-visual --batch-size 1  

Still encountering OOM, what other parameters can be adjusted?

navono, 2025-03-27:
with this scripts work on 4 4090D
```python
# SPDX-FileCopyrightText: Copyright (c) 2025 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
# SPDX-License-Identifier: Apache-2.0
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
# http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import os
import subprocess
import sys
from dataclasses import dataclass
from pathlib import Path

import torch
import tyro
from transformers import TrainingArguments

from gr00t.data.dataset import LeRobotSingleDataset
from gr00t.data.schema import EmbodimentTag
from gr00t.experiment.data_config import DATA_CONFIG_MAP
from gr00t.experiment.runner import TrainRunner
from gr00t.model.gr00t_n1 import GR00T_N1
from gr00t.utils.peft import get_lora_model

@dataclass
class Config:
    """Configuration for GR00T model fine-tuning."""

    # Dataset parameters
    dataset_path: str
    """Path to the dataset directory."""

    output_dir: str = "/tmp/gr00t"
    """Directory to save model checkpoints."""

    data_config: str = "gr1_arms_only"
    """Data configuration name from DATA_CONFIG_MAP."""

    # Training parameters
    batch_size: int = 16
    """Batch size per GPU for training."""

    max_steps: int = 10000
    """Maximum number of training steps."""

    num_gpus: int = 1
    """Number of GPUs to use for training."""

    save_steps: int = 500
    """Number of steps between saving checkpoints."""

    # Model parameters
    base_model_path: str = "nvidia/GR00T-N1-2B"
    """Path or HuggingFace model ID for the base model."""

    tune_llm: bool = False
    """Whether to fine-tune the language model backbone."""

    tune_visual: bool = True
    """Whether to fine-tune the vision tower."""
[... 172 lines cut ...]
                "torchrun",
                "--standalone",
                f"--nproc_per_node={config.num_gpus}",
                "--nnodes=1",  # default to 1 node for now
                str(script_path),
            ]

            # Convert config to command line arguments
            for key, value in vars(config).items():
                if isinstance(value, bool):
                    # For boolean values, use --flag or --no-flag format
                    if value:
                        cmd.append(f"--{key.replace('_', '-')}")
                    else:
                        cmd.append(f"--no-{key.replace('_', '-')}")
                else:
                    # For non-boolean values, use --key value format
                    cmd.append(f"--{key.replace('_', '-')}")
                    cmd.append(str(value))
            print("Running torchrun command: ", cmd)
            env = os.environ.copy()
            env["IS_TORCHRUN"] = "1"
            sys.exit(subprocess.run(cmd, env=env).returncode)

```

rrzhang139, 2025-03-28:
The script resolves the OOM error, but I also keep getting nan in my gradients.

youliangtan, 2025-04-18:
Im not getting NAN or OOM error when rerun the same script . Please reopen this issue if it isnt working again. Thanks!

closed 2025-04-18 by youliangtan

Nathaniel-Bubis, 2025-04-28:
> with this scripts work on 4 4090D
> 

4090D with 24 GB or the Chinese version with 48GB?

navono, 2025-05-06:
> > with this scripts work on 4 4090D
> 
> 4090D with 24 GB or the Chinese version with 48GB?

4090D with 24 GB

Wvirgil123, 2025-05-07:
> > > with this scripts work on 4 4090D
> > 
> > 
> > 4090D with 24 GB or the Chinese version with 48GB?
> 
> 4090D with 24 GB

@navono Does running the command have additional configuration? I use your script, then use the command below ,  4* 4090D with 24 GB ,result: OOM.
`python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 4 --max-steps 500 --output-dir 
/tmp/gr00t-1/finetuned-model --data-config gr1_arms_only  --batch_size 1`

navono, 2025-05-09:
> [@navono](https://github.com/navono) Does running the command have additional configuration? I use your script, then use the command below , 4* 4090D with 24 GB ,result: OOM. `python [...]

try `--no-tune-visual`

nbubis, 2025-05-09:
> try `--no-tune-visual`

That effectively means you cannot train on the 4090. Unless your embodiment is almost identical to the ones they trained on, it's unlikely freezing the visual layers will work.

navono, 2025-05-09:
You're right. In the end, I gave up training on the 4090 and rented an H20 instead

youliangtan, 2025-05-11:
use `LoRA` if compute is an issue. or multigpu setup e.g. 2 x 4090