NVIDIA/Isaac-GR00T #26: robot_sim.PickNPlace dataset replay in Sim Env and inference in the same  Sim Env

becauseofAI, 2025-03-20:
### 🚀 The feature, motivation and pitch

Can you provide the simulation environment for collecting robot_sim.PickNPlace dataset? In this way, we can replay robot_sim.PickNPlace data in the simulation environment. At the same time, we can do model inferring verification and result visualization in the simulation environment. It is not like this [tutorial](https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/1_gr00t_inference.ipynb) that can only run single-step inferring to get actions and cannot get environmental feedback of the actions.

youliangtan, 2025-03-20:
Thanks for raising this issue and I fully agree this "close loop sim evaluation" is missing. Internally we are doing a cleanup on our sim and hopefully it will get released soon.

Meanwhile, if you could easily integrate use gr00tn1 with the existing opensource sim, I recommend [libero](https://libero-project.github.io/main.html), [robocasa](https://robocasa.ai/), or others for posttraing then evaluation.

freestylerick, 2025-03-24:
Thanks; do you have a rough estimate on how long til this environment is released?

allisontw, 2025-07-04:
Hi, was wondering if there is any update?