NVIDIA/Isaac-GR00T #30: metadata.json is still needed

rrzhang139, 2025-03-20:
This is referring to this issue: https://github.com/NVIDIA/Isaac-GR00T/issues/28

The example you directed to me only obtains the modality.json needed for dataset. But I am referring to the metadata.json file needed for the models. I have been able to get the dataset stats and modalities for the strawberry dataset using `load_dataset.py`, but how can I consolidate it into the metadata.json file you had in the gr1 dataset? Thanks!

HuFY-dev, 2025-03-21:
I would suspect that you are trying to run the policy zero-shot on SO-100. Since SO-100 isn't in our pretraining, We do **not** have the metadata for it stored in our checkpoint. You'll need to finetune the models yourself before running the evaluation. 

LMK if this solves your problem.

rrzhang139, 2025-03-21:
I see. So if I understand correctly, we keep the `Gr00TPolicy` embodiment as `GR1` and then we load a dataset from a different embodiment (e.g `strawberry`) and call the `gr00t_finetune.py` script? Then it outputs the metadata.json?

HuFY-dev, 2025-03-21:
I think there are some misunderstanding here. Gr00tPolicy is a class for inference only. During fine-tuning, you'll specify the embodiment tag, and the trainer will save the matadata.json in the checkpoint. Also, our example always use the "new_embodiment" embodiment tag.

HuFY-dev, 2025-03-21:
Thanks for the detailed error information! We'll try to reproduce this issue and see if we can fix it.

rrzhang139, 2025-03-21:
https://github.com/NVIDIA/Isaac-GR00T/pull/39

HuFY-dev, 2025-03-21:
I see. The root cause is likely CUDA OOM. We are reviewing and testing that PR now and will merge that ASAP.

HuFY-dev, 2025-03-21:
FYI, we are using the Huggingface trainer with the following flags:

```python
    bf16=True,
    tf32=True,
```

This means it will perform most of the operations in bf16, except for certain numerically sensitive operations, and layer norm is one of the exceptions. In that case, it will use TensorFloat-32 which is faster than normal float32.

Also, I don't remember we have a `--bf16=True` flag in command line. Do you mean you set it in the training arguments?

rrzhang139, 2025-03-21:
Oh yes I mean I set it in the training args. Thanks!

youliangtan, 2025-04-07:
Reopen this issue if this is not resolved

closed 2025-04-07 by youliangtan