NVIDIA/Isaac-GR00T #32: Any Deploy this on Jetson AGX orin 64GB?

a3131919, 2025-03-21:
I am gonna try:)

johnnynunez, 2025-03-23:
https://github.com/dusty-nv/jetson-containers/tree/master/packages/robots/Isaac-GR00T yes for sure! use it

yult0821, 2025-06-06:
@a3131919  Hello, have you succussfully deployed gr00t on Jetson AGX orin 64GB?
@johnnynunez Hi! Following your instructions, I came to [jetson-containers](https://github.com/dusty-nv/jetson-containers/tree/master) packages/vla/isaac-gr00t/ trying to deploy gr00t N1 on jetson agx orin. When I try "autotag groot" in jetson teminal, it shows that there is no corresponding image yet. Then I choose to build it (follwing packages/vla/isaac-gr00t/Dockerfile? i guess), but some errors happen (see below). Is this a correct way? For the full log, please see the attachment "build_gr00t_image.log".

I also tried to directly follow the "pip install -e ." commond, it encounters the problem of installing decord.
> ERROR: Could not find a version that satisfies the requirement decord==0.6.0; platform_system != "Darwin" (from gr00t) (from versions: none)
> ERROR: No matching distribution found for decord==0.6.0; platform_system != "Darwin"

Further, I am wondering will you provide an image for gr00t suitable for jetson soon, like that for openvla in https://www.jetson-ai-lab.com/openvla.html#vla-architecture. Or is there another way to deploy gr00t N1 on jetson agx orin? 
Looking forward to your relpy. Thanks very much!

```
+ pip3 install --no-binary :all: psutil
Using pip 25.1.1 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
ERROR: Could not find a version that satisfies the requirement psutil (from versions: none)
ERROR: No matching distribution found for psutil
The command '/bin/sh -c $TMP/install.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/data/reComputer/jetson-containers/jetson_containers/tag.py", line 58, in <module>
    image = find_container(args.packages[0], prefer_sources=args.prefer, disable_sources=args.disable, user=args.
user, quiet=args.quiet)
  File "/data/reComputer/jetson-containers/jetson_containers/container.py", line 643, in find_container
    return build_container('', package) #, simulate=True)
  File "/data/reComputer/jetson-containers/jetson_containers/container.py", line 229, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host   --tag groot:r36.4-cu126-2
2.04-python   --file /data/reComputer/jetson-containers/packages/build/python/Dockerfile   --build-arg BASE_IMAGE
=groot:r36.4-cu126-22.04-cudnn   --build-arg PYTHON_VERSION="3.10"    /data/reComputer/jetson-containers/packages
/build/python 2>&1 | tee /data/reComputer/jetson-containers/logs/20250605_130819/build/groot_r36.4-cu126-22.04-py
thon.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
-- Error:  return code 1
```

[build_gr00t_image.log](https://github.com/user-attachments/files/20620932/build_gr00t_image.log)

closed 2025-06-11 by HuFY-dev