NVIDIA/Isaac-GR00T #38: CUDA OOM on smaller GPUs

karpenet, 2025-03-21:
### 🐛 Describe the bug

In `_load_model` method of `Gr00tPolicy` class, importing model weights with default `torch.float32` causes CUDA OOM and Flash Attention to not intialize correctly.

`model = GR00T_N1.from_pretrained(model_path)
        model.eval() `

```
2025-03-20 23:09:26.082123: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-03-20 23:09:26.082152: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-03-20 23:09:26.083281: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-03-20 23:09:26.089486: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-03-20 23:09:26.839494: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Fetching 6 files: 100%|██████████| 6/6 [02:44<00:00, 27.44s/it]
Loading pretrained dual brain from /home/jasonx/.cache/huggingface/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /home/jasonx/.cache/huggingface/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in SiglipVisionModel is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in LlamaForCausalLM is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in LlamaModel is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
/home/jasonx/Dropbox/repos/apple_pie/Isaac-GR00T/gr00t/model/action_head/cross_attention_dit.py:249: FutureWarning: Accessing config attribute `output_dim` directly via 'DiT' object attribute is deprecated. Please access 'output_dim' over 'DiT's config object instead, e.g. 'unet.config.output_dim'.
  self.proj_out_2 = nn.Linear(self.inner_dim, self.output_dim)
Total number of DiT parameters:  537803776
Tune action head projector: True
Tune action head diffusion model: True
Some weights of the model checkpoint at /home/jasonx/.cache/huggingface/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb were not used when initializing GR00T_N1: ['action_head.decode_layer.bias', 'action_head.decode_layer.weight']
- This IS expected if you are initializing GR00T_N1 from the checkpoint of a model trained on another task or with another architecture (e.g. initializing a BertForSequenceClassification model from a BertForPreTraining model).
- This IS NOT expected if you are initializing GR00T_N1 from the checkpoint of a model that you expect to be exactly identical (initializing a BertForSequenceClassification model from a BertForSequenceClassification model).
Tune action head projector: True
Tune action head diffusion model: True
```

### Versions

Python 3.10.16
certifi==2025.1.31
charset-normalizer==3.4.1
filelock==3.18.0
fsspec==2025.3.0
huggingface-hub==0.29.3
idna==3.10
inquirerpy==0.3.4
packaging==24.2
pfzy==0.3.4
prompt_toolkit==3.0.50
PyYAML==6.0.2
requests==2.32.3
tqdm==4.67.1
typing_extensions==4.12.2
urllib3==2.3.0
wcwidth==0.2.13

shra-one-11, 2025-03-21:
Noticed the default float32 usage could lead to OOM on mid-range GPUs, especially with Flash Attention 2.0’s reliance on float16/bfloat16. Curious—was the choice of float32 intentional to prioritize broader device stability, or could adjusting default precision to float16 (with fallback) help users mitigate this without manual tweaks? Would love to contribute if precision configurability is being considered!

HuFY-dev, 2025-03-21:
Hi @karpenet, thanks for pointing this out! It was our oversight not to test on GPUs with lower VRAM. We really appreciate your contribution and are currently testing your PR.

@shra-one-11 to answer your question, I think it's just a bug. We are always casting to BF16 during inference: https://github.com/NVIDIA/Isaac-GR00T/blob/7a48e15baf163366d721cd5581ab35dd6ca5563a/gr00t/model/policy.py#L182

PR #39 (2025-03-24): Fix load HF weights with BF16
Fix #38 
Changes proposed in this pull request:
- GROOT_N1 loads HuggingFace model by default torch.dtype, however the HF weights are BF16. This cause CUDA OOM issues on smaller GPUs (tested on 4060Ti).
- This PR fixes the bug by passing `torch.dtype="auto"` to `from_pretrained` method called on the GROOT_N1 class while loading the model. This lets dtype to be automatically derived from the model’s weights.
 
## Before submitting

- [x] I've read and followed all steps in the [Making a pull request](https://github.com/NVIDIA/Isaac-GR00T/blob/main/CONTRIBUTING.md#m [... cut]
PR #42 (2025-03-24): Always use BF16 for inference.

Fixes #38 

Changes proposed in this pull request:

- Always use BF16 for inference to save memory and avoid numeric overflows.

## Before submitting

- [x] I've read and followed all steps in the [Making a pull request](https://github.com/NVIDIA/Isaac-GR00T/blob/main/CONTRIBUTING.md#making-a-pull-request)
    section of the `CONTRIBUTING` docs.
- [ ] I've updated or added any relevant docstrings.
- [ ] If this PR fixes a bug, I've added a test that will fail without my fix.
- [ ] If this PR adds a new feature, I've added tests that sufficiently cover my new fun [... cut]
closed 2025-03-24 by youliangtan