NVIDIA/Isaac-GR00T #41: Crash when try to infer on nvidia orin nano

liangtea, 2025-03-21:
### 🚀 The feature, motivation and pitch

When i try to infer on nvidia orin nano, the system will crash and reboot. Anyone knows why? But everything goes well when i run this model on my X86-64 computer with RTX3060.  I think the problem is that flash-attn only support 16FP in Orin Nano.

HuFY-dev, 2025-03-21:
Hi! To help us better understand what’s going on, could you please share any error messages or logs you see on the Orin Nano? It would also be helpful to know the exact command you used to run the model.

liangtea, 2025-03-22:
> Hi! To help us better understand what’s going on, could you please share any error messages or logs you see on the Orin Nano? It would also be helpful to know the exact command you used to run the model.

**Absolutely OK！Here is my test code：**
```
import os
import torch
import gr00t

from gr00t.data.dataset import LeRobotSingleDataset
from gr00t.model.policy import Gr00tPolicy

# change the following paths
MODEL_PATH = "nvidia/GR00T-N1-2B"

# REPO_PATH is the path of the pip install gr00t repo and one level up
REPO_PATH = os.path.dirname(os.path.dirname(gr00t.__file__))
DATASET_PATH = os.path.join(REPO_PATH, "demo_data/robot_sim.PickNPlace")
EMBODIMENT_TAG = "gr1"

device = "cuda" if torch.cuda.is_available() else "cpu"

from gr00t.experiment.data_config import DATA_CONFIG_MAP

data_config = DATA_CONFIG_MAP["gr1_arms_only"]
modality_config = data_config.modality_config()
modality_transform = data_config.transform()

policy = Gr00tPolicy(
    model_path=MODEL_PATH,
    embodiment_tag=EMBODIMENT_TAG,
    modality_config=modality_config,
    modality_transform=modality_transform,
    device=device,
)

# print out the policy model architecture
print(policy.model)

import numpy as np

modality_config = policy.modality_config

print(modality_config.keys())

for key, value in modality_config.items():
    if isinstance(value, np.ndarray):
        print(key, value.shape)
    else:
        print(key, value)

# Create the dataset
dataset = LeRobotSingleDataset(
    dataset_path=DATASET_PATH,
    modality_configs=modality_config,
    video_backend="decord",
    video_backend_kwargs=None,
    transforms=None,  # We'll handle transforms separately through the policy
    embodiment_tag=EMBODIMENT_TAG,
)

import numpy as np

step_data = dataset[0]

print(step_data)

predicted_action = policy.get_action(step_data)
for key, value in predicted_action.items():
    print(key, value.shape)
```
[... 3 lines cut ...]
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /root/.cache/huggingface/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in SiglipVisionModel is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in LlamaForCausalLM is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
Flash Attention 2.0 only supports torch.float16 and torch.bfloat16 dtypes, but the current dype in LlamaModel is torch.float32. You should run training or inference using Automatic Mixed-Precision via the `with torch.autocast(device_type='torch_device'):` decorator, or load the model with the `torch_dtype` argument. Example: `model = AutoModel.from_pretrained("openai/whisper-tiny", attn_implementation="flash_attention_2", torch_dtype=torch.float16)`
/mnt/nvme/Isaac-GR00T/gr00t/model/action_head/cross_attention_dit.py:249: FutureWarning: Accessing config attribute `output_dim` directly via 'DiT' object attribute is deprecated. Please access 'output_dim' over 'DiT's config object instead, e.g. 'unet.config.output_dim'.
  self.proj_out_2 = nn.Linear(self.inner_dim, self.output_dim)
Total number of DiT parameters:  537803776
Tune action head projector: True
Tune action head diffusion model: True
Some weights of the model checkpoint at /root/.cache/huggingface/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb were not used when initializing GR00T_N1: ['action_head.decode_layer.bias', 'action_head.decode_layer.weight']
- This IS expected if you are initializing GR00T_N1 from the checkpoint of a model trained on another task or with another architecture (e.g. initializing a BertForSequenceClassification model from a BertForPreTraining model).
- This IS NOT expected if you are initializing GR00T_N1 from the checkpoint of a model that you expect to be exactly identical (initializing a BertForSequenceClassification model from a BertForSequenceClassification model).
Tune action head projector: True
Tune action head diffusion model: True
``` 
Here is my hardware status：
03-22-2025 09:53:28 RAM 7418/7620MB (lfb 6x4MB) SWAP 5735/16384MB (cached 244MB) CPU [11%@729,17%@729,1%@729,1%@729,50%@1036,7%@1036] EMC_FREQ 2%@2133 GR3D_FREQ 2%@[305] NVDEC off NVJPG off NVJPG1 off VIC off OFA off APE 200 cpu@47.25C soc2@46.562C soc0@46.937C gpu@48.093C tj@48.093C soc1@47.031C VDD_IN 5467mW/5065mW VDD_CPU_GPU_CV 909mW/775mW VDD_SOC 1505mW/1442mW
03-22-2025 09:53:29 RAM 7377/7620MB (lfb 6x4MB) SWAP 5704/16384MB (cached 115MB) CPU [6%@729,7%@729,3%@729,25%@729,32%@1113,6%@1113] EMC_FREQ 1%@2133 GR3D_FREQ 2%@[305] NVDEC off NVJPG off NVJPG1 off VIC off OFA off APE 200 cpu@46.937C soc2@46.562C soc0@46.843C [... cut ...] 8C tj@48C soc1@46.968C VDD_IN 5325mW/5066mW VDD_CPU_GPU_CV 793mW/775mW VDD_SOC 1470mW/1442mW
03-22-2025 09:53:30 RAM 7286/7620MB (lfb 6x4MB) SWAP 5683/16384MB (cached 3MB) CPU [8%@729,39%@729,1%@729,2%@729,29%@729,0%@729] EMC_FREQ 1%@2133 GR3D_FREQ 0%@[305] NVDEC off NVJPG off NVJPG1 off VIC off OFA off APE 200 cpu@47.125C soc2@46.562C soc0@46.906C gpu@48.062C tj@48.062C soc1@47.031C VDD_IN 5484mW/5068mW VDD_CPU_GPU_CV 872mW/775mW VDD_SOC 1468mW/1442mW

after that， My Orin nano crash and reboot!

ATHARV-CHAUDHAR, 2025-03-22:
👉 Quick Recap of the Problem:
You’re trying to run inference on the NVIDIA Orin Nano, but your system crashes or reboots. It runs fine on your RTX 3060, but Orin Nano seems to struggle. The reason could be hardware limitations, unsupported FP16 precision, or software compatibility issues. Mine was  unsupported FP16 precision. It was solved and while solving this I got to learn this many things, so check it out if you find it usefull.

👉 Step-by-Step, Solution:
1. Switch to FP32 Precision
The Orin Nano might not handle FP16 well, or it may be unstable.

Libraries like flash-attn try to use FP16 for performance, but it can cause crashes on this hardware.

Solution: Change your model and inference settings to use float32 (FP32) precision.
In PyTorch, make sure you're doing something like:

`model = model.to(torch.float32)`
Or set dtype=torch.float32 wherever needed.`

2. Reduce the Batch Size
The memory on Orin Nano is very limited compared to RTX 3060.

Large batch sizes can easily cause memory overflow and system crashes.

Solution: Start with a batch size of 1. If it runs stable, maybe try 2, but keep it small.

3. Optimize Your Model
Running large models on edge devices without optimization is not ideal.

What you can do:

Quantize your model (try converting it to INT8 or keep it on FP32 if that’s more stable).

Use model pruning to reduce unnecessary parameters.

Leverage TensorRT optimization — NVIDIA provides this for Jetson devices to run models faster and lighter.

4. Use Jetson-Specific Software
The software environment on Jetson devices is very specific.

You must use versions of PyTorch, CUDA, cuDNN, and other libraries that are built for Jetson/ARM architectures.

Solution: Install everything through the official JetPack SDK or from NVIDIA's recommended sources for Jetson platforms.

5. Check System Logs for Clues
After a crash, check logs:

Run dmesg to see kernel and GPU-related errors.

Check /var/log/syslog or /var/log/kern.log to see if there are "Out of Memory" or GPU fault messages.

This can help confirm whether it's a memory issue or something else.

6. Monitor Resource Usage in Real-Time
Run:

bash

`sudo tegrastats
This shows CPU, GPU, and memory usage in real time.`

Watch it during inference to see where the system is hitting its limits.

🔎 In short:
Force your inference to use FP32 → Reduce batch size to 1 or 2 → Optimize your model with TensorRT, pruning, and possibly quantization → Only use Jetson-compatible builds → Check system logs → Monitor with tegrastats to know exactly where the bottleneck is.

liangtea, 2025-03-24:
> 👉 Quick Recap of the Problem: You’re trying to run inference on the NVIDIA Orin Nano, but your system crashes or reboots. It runs fine on your RTX 3060, but Orin Nano seems to struggle. The reason [...]

Thanks for your reply. But i found that the problem is that GPU memory is not enough. This model will occupy at least 10GB memory. But INVIDIA Jetson Orin Nano only has 8GB memory and it will be crashed when loading such a big model. So as you said above , 'Optimize your model with TensorRT, pruning, and possibly quantization' may help.

Hao-yang-Wang, 2025-03-25:
Hi, I'm also trying to infer. I use 3080ti, but it also shows the GPU memory is not enough error. Do you use only one 3060 to do the inference?

liangtea, 2025-03-25:
> Hi, I'm also trying to infer. I use 3080ti, but it also shows the GPU memory is not enough error. Do you use only one 3060 to do the inference? 

Sorry. I typed a wrong mode in my issue. Actually this model needs approximately 11GB GPU RAM.

ATHARV-CHAUDHAR, 2025-03-30:
> Hi, I'm also trying to infer. I use 3080ti, but it also shows the GPU memory is not enough error. Do you use only one 3060 to do the inference?

It depends on the model size and optimization techniques used. A 3080 Ti has 12GB VRAM, so if you're running the model without optimizations, it might still run out of memory.

I used a single RTX 3060 (12GB) for inference, to run the model, but on conditions applied:

Quantization (8-bit or 4-bit using bitsandbytes or GPTQ) is applied.

Offloading some layers to CPU using accelerate or DeepSpeed ZeRO Offload.

TensorRT optimizations like INT8 calibration and sparsity are used.

If you're facing out-of-memory (OOM) issues even on a 3080 Ti, I recommend checking the batch size, enabling gradient checkpointing, or using FP16 precision to reduce memory consumption.

Let me know what settings you're using, and I can help debug further! 🚀

liangtea, 2025-04-01:
> It depends on the model size and optimization techniques used. A 3080 Ti has 12GB VRAM, so if you're running the model without optimizations, it might still run out of memory.
>
> I used a single RTX [...]

Hello. Could you upload you code here? So everyone can learn to optimize this model. Thanks.

Jerryisqx, 2025-04-02:
> It depends on the model size and optimization techniques used. A 3080 Ti has 12GB VRAM, so if you're running the model without optimizations, it might still run out of memory.
>
> I used a single RTX [...]

Hey, I would like to know can the bitandbytes quantization can be applied directly in the 'from_pretrained' method in the 'Isaac-GR00T/gr00t/model/gr00t_n1.py' ?