NVIDIA/Isaac-GR00T #47: How much VRAM is required to run inference?

Hao-yang-Wang, 2025-03-22:
I'm using the 3080ti to run the inference example, but it returns CUDA out of memory. What is the minimum amount of VRAM required to run this model?

ATHARV-CHAUDHAR, 2025-03-22:
**Minimum VRAM requirement:**

- For full precision (FP16/FP32): at least 24GB VRAM.

- With quantization (4-bit/8-bit): can attempt on 12–16GB, but still might need model parallelism or offloading.
 
- Context length and batch size tuning also plays a huge role.
 
- Large models like this often require A100 or at least 3090/4090 with VRAM > 24GB.

You cannot run this large GR00T model on 12GB VRAM without extreme optimization or quantization.
NVIDIA’s large foundational robotics models generally need 24GB VRAM minimum.
If only 12GB is available, quantization + batch size 1 + short sequence length might still fail — best to use A100, or 3090/4090, or run inference through Triton Inference Server with CPU/GPU hybrid memory management.

jackylee1, 2025-03-24:
Can I run it on windows platform?

ATHARV-CHAUDHAR, 2025-03-30:
> Can I run it on windows platform?

Yes, but there are several limitations, and the setup needs to be done properly.

Problems you may encounter on Windows:
✅ CUDA & Driver Issues: The CUDA stack on Windows is not as efficient as on Linux, especially for PyTorch and TensorRT inference.
✅ VRAM Usage: Background processes on Windows also consume GPU memory, so the available VRAM is lower compared to Linux.
✅ Model Parallelism Issues: PyTorch’s FSDP (Fully Sharded Data Parallelism) and DeepSpeed ZeRO work best on Linux but may not perform well on Windows.

youliangtan, 2025-04-18:
Since this is a 2B model, with bfloat16, it would just require approx `4.9GB` gpu vram, tested on my local compute

closed 2025-04-18 by youliangtan