NVIDIA/Isaac-GR00T #48: How to merge multiple datasets for finetuning?

johnnynunez, 2025-03-23:
### 🐛 Describe the bug

Put multiple dataset for training

### Versions

last commit

youliangtan, 2025-04-07:
We can release our MultiDatasetClass method for this usage.

Meanwhile check this on lerobot: https://github.com/huggingface/lerobot/issues/533

closed 2025-04-18 by qiwang-nvidia

zhuhui123, 2025-05-13:
> We can release our MultiDatasetClass method for this usage.
> 
> Meanwhile check this on lerobot: [huggingface/lerobot#533](https://github.com/huggingface/lerobot/issues/533)

When will it be released? I haven't found the MultiDatasetClass method yet.

zhuhui123, 2025-06-03:
Sorry, I haven't received a reference about this yet. Can you help me confirm it again, or send it to my email?

youliangtan, 2025-06-03:
@zhuhui123 We will make a new version update next week. That will include `MultiDataset` loading