NVIDIA/Isaac-GR00T #49: Inference: System 2 runs at 10Hz and System1 generates actions at 120Hz

Yochengliu, 2025-03-24:
### 📚 The doc issue

The paper says: 
The System 2 reasoning module at 10Hz on an NVIDIA L40 GPU. Subsequently, a Diffusion Transformer serves
as the System 1 action module. It generates closed-loop motor actions at a higher frequency (120Hz).

So, it seems that the System 2 and System1 run asynchronously as you claim above, at inference stage. But I can not find the corresponding documentation/implementation in this repo. Please clarify this part, thanks.

### Suggest a potential alternative/fix

Clarify what strategy the GR00T uses for System 2 and System1 at inference stage.

Catchip, 2025-04-14:
I've progressively grasped how the asynchrony between System 1 and System 2 is implemented. While System 2 captures the current frame image and processes reasoning, System 1 receives System 2's inference results, outputs an action sequence of length action horizon, and directly executes the actions. 

```python
img = robot.get_current_img()
view_img(img)
state = robot.get_current_state()
action = client.get_action(img, state)
start_time = time.time()
for i in range(ACTION_HORIZON):
    concat_action = np.concatenate(
        [np.atleast_1d(action[f"action.{key}"][i]) for key in MODALITY_KEYS],
        axis=0,
    )
    assert concat_action.shape == (6,), concat_action.shape
    robot.set_target_state(torch.from_numpy(concat_action))
```

Notably, in getting_started.py, the horizon is set to 12 by default – which likely explains why System 1 operates at 12 times the frequency of System 2.

```python
parser.add_argument("--action_horizon", type=int, default=12)

### other codes

ACTION_HORIZON = (
    args.action_horizon
)  # we will execute only some actions from the action_chunk of 16
```

Yochengliu, 2025-04-14:
@Catchip  That's not an 'asynchronous' running like Figure Helix, just a common practice in robot control using VLA model. I think maybe the claim in the paper is somewhat biased.

closed 2025-04-14 by Yochengliu

Catchip, 2025-04-14:
@Yochengliu  I noticed the paper doesn't explicitly mention implementing an 'asynchronous' mechanism. The descriptions of S1 and S2 initially led me to mistakenly assume it employed an asynchronous architecture similar to Helix or HiRT.
As you  pointed out, this does not actually constitute asynchronous operation.

EricYang3721, 2025-05-27:
Thanks for details above. I found the same issue. This is not the `asynchronous` as I expected.