NVIDIA/Isaac-GR00T #52: illegal memory error on 3090 24GB with Lora finetuning

rrzhang139, 2025-03-24:
Hello, I have tried using this [lora finetuning script](https://github.com/neil-tan/Isaac-GR00T/blob/peft/scripts/gr00t_finetune.py) to efficiently train my model, but I keep getting illegal memory errors that may be related to OOM but I am not too sure because it shows only 8GB is loaded when my debugger stepped into the illegal memory error at [trainer.train()](https://github.com/NVIDIA/Isaac-GR00T/blob/main/gr00t/experiment/runner.py#L164)

[image]

Here is my lora model showing it's finetuning on 11% of the total params:

`trainable params: 2,568,192 || all params: 2,192,584,944 || trainable%: 0.1171`

Is this due to OOM or is there something else going on here? 

Related: https://github.com/NVIDIA/Isaac-GR00T/issues/30

neil-tan, 2025-03-27:
Hi @rrzhang139,
  Can you provide the complete command you are running the fine-tune script with? As well as where you set the breakpoint and your system information?

neil-tan, 2025-03-27:
@rrzhang139 
Try decreasing the batch size:
`python scripts/gr00t_finetune.py  --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 1 --lora_rank 64  --lora_alpha 128  --batch-size 24`

rrzhang139, 2025-03-27:
It worked! Thank you. I think there is something wrong with the batch size when I set it to 8. batch size 32 also does not work for me.

closed 2025-03-27 by rrzhang139