NVIDIA/Isaac-GR00T #53: Running Isaac-GR00T in Isaac Sim

Merai-Nitish, 2025-03-25:
### 📚 The doc issue

Is there any way we can install Isaac-GR00T in Isaac Sim and see things in action like watch a Unitree G1 humanoid locomotion and pick and place.

aixiamomo, 2025-03-27:
I also want to know how to do it 🤔

wuxiaoqiang12, 2025-03-28:
I also want to know, I have launched Unitree go2_omniverse, I want integrate GR00T into it to manipulate G1 robot to interact with env.

mkygogo, 2025-03-29:
I also want to know how to do it 🤔

youliangtan, 2025-04-18:
Thanks for all the interests in this. There are many similar issues related to run this in a sim eval environoment. The TLDR is, we will soon release a suite of sim envs. Stay tuned.

https://github.com/NVIDIA/Isaac-GR00T/issues/26#issuecomment-2740917368

> Thanks for raising this issue and I fully agree this "close loop sim evaluation" is missing. Internally we are doing a cleanup on our sim and hopefully it will get released soon.

> Meanwhile, if you could easily integrate use gr00tn1 with the existing opensource sim, I recommend [libero](https://libero-project.github.io/main.html), [robocasa](https://robocasa.ai/), or others for posttraing then evaluation.

RN-Tesseract, 2025-04-22:
Hi @qiwang-nvidia  and @youliangtan 
Looking forward to this to run pre-trained models on either native GR1 or other humanoid in Isaac sim.
Are you guys also working on creating the environment for specific tasks (Pick and place / sorting) ? and also a Text interface to provide inputs to Isaac sim. ?

zz7379, 2025-04-29:
on this too!

Welnic, 2025-05-02:
Interested in this as well!

AgentEXPL, 2025-07-25:
I am following up to see if there are any updates.