NVIDIA/Isaac-GR00T #54: Basic usage questions

GeraltRivskiy, 2025-03-25:
Hello!
Thanks for your work! I'd like to ask you for some advice regarding the following points:

1) If I understand correctly, there's no practical use for the pretrained model directly on custom robots. Therefore, I need to gather a dataset specific to my robot and then fine-tune the model, is that right?

2) In the getting_started section, you fine-tuned the model for the SO100 arm using a dataset of 40 episodes, and it gave decent results. Would you say 40 episodes are generally enough for a rough fine-tuning of a custom robot, or should it typically be 10x or even 100x more?

3) How well does the model generalize? For example, if the task involves placing an object on a plate or in a jar, would fine-tuning the model with plates also allow it to perform well with jars? Basically, I'm curious whether fine-tuning is required for every single task, or if it primarily adapts the model to a new embodiment, after which it can successfully perform tasks similar to those it was originally trained on.

4) Similarly, regarding different objects involved in the same task—like placing bananas or strawberries on a plate—would I need separate datasets for each object, or could the model generalize well to strawberries if fine-tuned only with bananas?

5) How does the model handle variations in task prompts? Specifically, how robust is it to synonyms, typos, grammatical errors, and similar variations in prompts describing the same task?

I'm asking these questions mainly to understand what steps I need to take to make this model work effectively with my robot in undefined environments.

ShubhamDindorkar, 2025-03-27:
1. If I understand correctly, there's no practical use for the pretrained model directly on custom robots. Therefore, I need to gather a dataset specific to my robot and then fine-tune the model, is that right?

-  No, pretrained models are typically trained on standardized datasets and robot configurations. You need to fine-tune the model with your own dataset that reflects your specific robot’s dynamics, sensors, and actuators. Think of GR00T as a "smart brain" that needs to learn how to use your robot’s "body."

2. In the getting_started section, you fine-tuned the model for the SO100 arm using a dataset of 40 episodes, and it gave decent results. Would you say 40 episodes are generally enough for a rough fine-tuning of a custom robot, or should it typically be 10x or even 100x more?

-  In the SO100 example, 40 episodes worked decently, but that was because:
The pretrained model already had a solid base.
But, Dynamic tasks (e.g., folding clothes) need more data. So, start with 40–100 episodes, then iteratively add data where the model fails.

3. How well does the model generalize? For example, if the task involves placing an object on a plate or in a jar, would fine-tuning the model with plates also allow it to perform well with jars? Basically, I'm curious whether fine-tuning is required for every single task, or if it primarily adapts the model to a new embodiment, after which it can successfully perform tasks similar to those it was originally trained on.

-  If fine-tuned on placing objects on plates, the model might transfer to jars if the motion is similar. However, jars (deep, narrow) vs. plates (flat) may need additional data due to differing collision dynamics. If the action and object physics are similar, the model may generalize. Otherwise, more training is needed.

4. Similarly, regarding different objects involved in the same task—like placing bananas or strawberries on a plate—would I need separate datasets for each object, or could the model generalize well to strawberries if fine-tuned only with bananas?

-  If both objects share physical properties (size, weight, material), the model can generalize reasonably well.
If they differ significantly (e.g., a banana is soft and curved, while a strawberry is firm and round), the model might struggle.
Try fine-tuning with a dataset containing a mix of similar objects to improve generalization.

5. How does the model handle variations in task prompts? Specifically, how robust is it to synonyms, typos, grammatical errors, and similar variations in prompts describing the same task?

- Pretrained language models are often trained on diverse text, so they handle minor variations well. Simple errors are usually tolerated, but complex phrasing may confuse the model. For safety-critical tasks, avoid ambiguous prompts. 

I hope I have addressed most of your questions. These are my handwritten notes, which I took while learning about this topic. I hope they prove helpful. 😊

GeraltRivskiy, 2025-03-28:
@ShubhamDindorkar yes, thanks a lot, your answer is really helpful and allows me to consider about how much time this task could take!

closed 2025-03-28 by GeraltRivskiy