NVIDIA/Isaac-GR00T #56: tune-llm has no effect on trainable params

cagbal, 2025-03-26:
### 🐛 Describe the bug

tune-llm and no-tune-llm returns the same number of params. Is this expected?

checked it with the addition below to the scripts/gr00t_finetune.py:

```
trainable_params = sum(p.numel() for p in model.parameters() if p.requires_grad)
print(f"Total Trainable Parameters: {trainable_params:,}") 

total_params = sum(p.numel() for p in model.parameters())
print(f"Total Parameters:         {total_params:,}")
print(f"Non-Trainable Parameters: {total_params - trainable_params:,}") 
```

```
python scripts/gr00t_finetune.py \
   --dataset-path /workspace/Isaac-GR00T/demo_data/test_data \
   --num-gpus 1 \
   --output-dir /workspace/checkpoints  \
   --max-steps 10000 \
   --data-config so100 \
   --video-backend torchvision_av \
   --no-tune-llm \
   --tune-visual \
   --tune-diffusion-model \
   --tune-projector
```

### Versions

Python 3.11.10
absl-py==2.2.0
accelerate==1.2.1
aiohappyeyeballs==2.6.1
aiohttp==3.11.14
aiosignal==1.3.2
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
anyio==4.6.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
astunparse==1.6.3
async-lru==2.0.4
attrs==24.2.0
av==12.3.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.1.0
blessings==1.7
blinker==1.4
cachetools==5.5.2
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
click==8.1.8
cloudpickle==3.1.1
comm==0.2.2
contourpy==1.3.1
cramjam==2.9.1
cryptography==3.4.8
cycler==0.12.1
datasets==3.4.1
dbus-python==1.2.18
debugpy==1.8.5
decorator==5.1.1
decord==0.6.0
defusedxml==0.7.1
diffusers==0.30.2
dill==0.3.8
distro==1.7.0
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.16
einops==0.8.1
entrypoints==0.4
eval_type_backport==0.2.2
executing==2.1.0
Farama-Notifications==0.0.4
fastjsonschema==2.20.0
fastparquet==2024.11.0
filelock==3.13.1
flash-attn==2.7.1.post4
flatbuffers==25.2.10
fonttools==4.56.0
fqdn==1.5.1
frozenlist==1.5.0
fsspec==2024.2.0
fvcore==0.1.5.post20221221
gast==0.6.0
gitdb==4.0.12
GitPython==3.1.44
google-auth==2.38.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
-e git+https://github.com/NVIDIA/Isaac-GR00T@153ee6fee23d4c3bf52cb4b520ddd258f98d017b#egg=gr00t
grpcio==1.71.0
gymnasium==1.0.0
h11==0.14.0
h5py==3.12.1
httpcore==1.0.5
httplib2==0.20.2
httpx==0.27.2
huggingface-hub==0.29.3
hydra-core==1.3.2
idna==3.10
imageio==2.34.2
importlib-metadata==4.6.4
iniconfig==2.1.0
inquirerpy==0.3.4
iopath==0.1.10
ipykernel==6.29.5
ipython==8.27.0
ipython-genutils==0.2.0
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.1
jeepney==0.7.1
Jinja2==3.1.3
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-archive==3.4.0
jupyter-events==0.10.0
jupyter-highlight-selected-word==0.2.0
jupyter-lsp==2.2.5
jupyter_client==7.4.9
jupyter_contrib_core==0.4.2
jupyter_contrib_nbextensions==0.7.0
jupyter_core==5.7.2
jupyter_nbextensions_configurator==0.6.4
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.5
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==3.0.13
keras==2.15.0
keyring==23.5.0
kiwisolver==1.4.8
kornia==0.7.4
kornia_rs==0.1.8
launchpadlib==1.10.16
lazr.restfulclient==0.14.4
lazr.uri==1.0.6
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
lxml==5.3.0
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.10.0
matplotlib-inline==0.1.7
mdurl==0.1.2
mistune==3.0.2
ml-dtypes==0.2.0
more-itertools==8.10.0
mpmath==1.3.0
msgpack==1.1.0
multidict==6.2.0
multiprocess==0.70.16
nbclassic==1.1.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
networkx==3.2.1
notebook==6.5.5
notebook_shim==0.2.4
numba==0.61.0
numpy==1.26.3
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.2.0
omegaconf==2.3.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
opt_einsum==3.4.0
overrides==7.7.0
packaging==24.1
pandas==2.2.3
pandocfilters==1.5.1
parso==0.8.4
peft==0.14.0
pettingzoo==1.24.3
pexpect==4.9.0
pfzy==0.3.4
pillow==10.2.0
pipablepytorch3d==0.7.6
platformdirs==4.3.6
pluggy==1.5.0
portalocker==3.1.1
prometheus_client==0.21.0
prompt_toolkit==3.0.47
propcache==0.3.1
protobuf==3.20.3
psutil==6.0.0
ptyprocess==0.7.0
pure_eval==0.2.3
pyarrow==14.0.1
pyasn1==0.6.1
pyasn1_modules==0.4.1
pyav==14.2.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
Pygments==2.18.0
PyGObject==3.42.1
PyJWT==2.3.0
pyparsing==2.4.7
pytest==8.3.5
python-apt==2.4.0+ubuntu4
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytz==2025.2
PyYAML==6.0.2
pyzmq==24.0.1
ray==2.40.0
referencing==0.35.1
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.4
rpds-py==0.20.0
rsa==4.9
safetensors==0.5.3
scikit-image==0.25.2
scipy==1.15.2
SecretStorage==3.3.1
Send2Trash==1.8.3
sentry-sdk==2.24.1
setproctitle==1.3.5
shtab==1.7.1
six==1.16.0
smmap==5.0.2
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.5.0
terminado==0.18.1
tianshou==0.5.1
tifffile==2025.3.13
timm==1.0.14
tinycss2==1.3.0
tokenizers==0.20.3
torch==2.5.1
torchaudio==2.4.1+cu124
torchvision==0.20.1
tornado==6.4.1
tqdm==4.67.1
traitlets==5.14.3
transformers==4.45.2
triton==3.1.0
typeguard==4.4.2
types-python-dateutil==2.9.0.20240906
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
uri-template==1.3.0
urllib3==2.2.3
wadllib==1.3.6
wandb==0.18.0
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0
Werkzeug==3.1.3
widgetsnbextension==4.0.13
wrapt==1.14.1
xxhash==3.5.0
yacs==0.1.8
yarl==1.18.3
zipp==1.0.0
zmq==0.0.0

cagbal, 2025-03-27:
Hi, thank you for your response. I  believe in this case lora is not active.

shenyanqing1105, 2025-05-15:
> Hi, thank you for your response. I believe in this case lora is not active.

same question! Have you solved this? @cagbal 
I need help!! @youliangtan @HuFY-dev

cagbal, 2025-05-15:
No, havent worked on it yet.

lakomchik, 2025-05-15:
I also observed that `tune_llm` had no effect, so I updated the method in `gr00t/model/backbone/eagle_backbone.py`:

---

**Original (lines 160–167):**
```python
def set_trainable_parameters(self, tune_llm: bool, tune_visual: bool):
    self.tune_llm = tune_llm
    self.tune_visual = tune_visual
    if not tune_llm:
        self.model.language_model.requires_grad_(False)
    if not tune_visual:
        self.model.vision_model.requires_grad_(False)
        self.model.mlp1.requires_grad_(False)

---
**Modified implementation:**
```python
def set_trainable_parameters(self, tune_llm: bool, tune_visual: bool):
    self.tune_llm = tune_llm
    self.tune_visual = tune_visual

    if not tune_llm:
        self.model.language_model.requires_grad_(False)
    else:
        self.model.language_model.requires_grad_(True)

    if not tune_visual:
        self.model.vision_model.requires_grad_(False)
        self.model.mlp1.requires_grad_(False)
    else:
        self.model.vision_model.requires_grad_(True)
        self.model.mlp1.requires_grad_(True)

shenyanqing1105, 2025-05-15:
> I also observed that `tune_llm` had no effect, so I updated the method in `gr00t/model/backbone/eagle_backbone.py`:
> 
> **Original (lines 160–167):**
> 
> def set_trainable_parameters(self, tune_llm: bool, tune_visual: bool):
>     self.tune_llm = tune_llm
>     self.tune_visual = tune_visual
>     if not tune_llm:
>         self.model.language_model.requires_grad_(False)
>     if not tune_visual:
>         self.model.vision_model.requires_grad_(False)
>         self.model.mlp1.requires_grad_(False)
> **Modified implementation:**
> 
> def set_trainable_parameters(self, tune_llm: bool, tune_visual: bool):
>     self.tune_llm = tune_llm
>     self.tune_visual = tune_visual
> 
>     if not tune_llm:
>         self.model.language_model.requires_grad_(False)
>     else:
>         self.model.language_model.requires_grad_(True)
> 
>     if not tune_visual:
>         self.model.vision_model.requires_grad_(False)
>         self.model.mlp1.requires_grad_(False)
>     else:
>         self.model.vision_model.requires_grad_(True)
>         self.model.mlp1.requires_grad_(True)

It works. I thought default is True haha...

HuFY-dev, 2025-06-11:
Solved in 1.5 release:

https://github.com/NVIDIA/Isaac-GR00T/blob/17a77ebf646cf13460cdbc8f49f9ec7d0d63bcb1/gr00t/model/backbone/eagle_backbone.py#L68-L69

closed 2025-06-11 by HuFY-dev