NVIDIA/Isaac-GR00T #57: Missing q01 and q99 in unitree dataset

Rohith-Scalers, 2025-03-26:
### 🐛 Describe the bug

unitree block dataset is missing q01 and q99 , are thee values required or optional ? if required how to generate these values 
```
python3 scripts/load_dataset.py --data_path /home/metrum/rohith-groot/G1_BlockStacking_Dataset --embodiment_tag new_embodiment
/home/metrum/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
  warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
/home/metrum/.local/lib/python3.10/site-packages/albumentations/__init__.py:13: UserWarning: A new version of Albumentations is available: 2.0.5 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
("Valid modality_keys for debugging:: {'state': ['state.left_arm', "
 "'state.right_arm', 'state.left_hand', 'state.right_hand'], 'action': "
 "['action.left_arm', 'action.right_arm', 'action.left_hand', "
 "'action.right_hand'], 'video': ['video.cam_right_high', "
 "'video.cam_left_wrist', 'video.cam_right_wrist'], 'annotation': "
 "['annotation.human.task_description']} \n")
state_modality_keys: ['state.left_arm', 'state.right_arm', 'state.left_hand', 'state.right_hand']
action_modality_keys: ['action.left_arm', 'action.right_arm', 'action.left_hand', 'action.right_hand']
Failed to load dataset statistics: 2 validation errors for DatasetStatisticalValues
q01
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
q99
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
Calculating dataset statistics for G1_BlockStacking_Dataset
Collecting all parquet files...:   0%|                                                                                                       | 0/301 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 315, in _get_metadata
    DatasetStatisticalValues.model_validate(stat)
  File "/home/metrum/.local/lib/python3.10/site-packages/pydantic/main.py", line 627, in model_validate
    return cls.__pydantic_validator__.validate_python(
pydantic_core._pydantic_core.ValidationError: 2 validation errors for DatasetStatisticalValues
q01
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
q99
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 162, in <module>
    load_dataset(args.data_path, args.embodiment_tag, args.video_backend)
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 97, in load_dataset
    dataset = LeRobotSingleDataset(
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 130, in __init__
    self._metadata = self._get_metadata(EmbodimentTag(self.tag))
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 321, in _get_metadata
    le_statistics = calculate_dataset_statistics(parquet_files)
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 55, in calculate_dataset_statistics
    parquet_data = pd.read_parquet(parquet_path)
  File "/home/metrum/.local/lib/python3.10/site-packages/pandas/io/parquet.py", line 667, in read_parquet
    return impl.read(
  File "/home/metrum/.local/lib/python3.10/site-packages/pandas/io/parquet.py", line 274, in read
    pa_table = self.api.parquet.read_table(
  File "/home/metrum/.local/lib/python3.10/site-packages/pyarrow/parquet/core.py", line 2956, in read_table
    dataset = _ParquetDatasetV2(
  File "/home/metrum/.local/lib/python3.10/site-packages/pyarrow/parquet/core.py", line 2496, in __init__
    [fragment], schema=schema or fragment.physical_schema,
  File "pyarrow/_dataset.pyx", line 1358, in pyarrow._dataset.Fragment.physical_schema.__get__
  File "pyarrow/error.pxi", line 154, in pyarrow.lib.pyarrow_internal_check_status
  File "pyarrow/error.pxi", line 91, in pyarrow.lib.check_status
pyarrow.lib.ArrowInvalid: Could not open Parquet input source '<Buffer>': Parquet magic bytes not found in footer. Either the file is corrupted or this is not a parquet file.
metrum@precision-7960:~/rohith-groot/Isaac-GR00T$ python3 scripts/load_dataset.py --data_path /home/metrum/rohith-groot/G1_BlockStacking_Dataset/ --embodiment_tag new_embodiment
/home/metrum/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
  warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
^[[200~python3 scripts/load_dataset.py --data_path /home/metrum/rohith-groot/G1_BlockStacking_Dataset/ --embodiment_tag new_embodiment^[[201~/home/metrum/.local/lib/python3.10/site-packages/albumentations/__init__.py:13: UserWarning: A new version of Albumentations is available: 2.0.5 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
("Valid modality_keys for debugging:: {'state': ['state.left_arm', "
 "'state.right_arm', 'state.left_hand', 'state.right_hand'], 'action': "
 "['action.left_arm', 'action.right_arm', 'action.left_hand', "
 "'action.right_hand'], 'video': ['video.cam_right_high', "
 "'video.cam_left_wrist', 'video.cam_right_wrist'], 'annotation': "
 "['annotation.human.task_description']} \n")
state_modality_keys: ['state.left_arm', 'state.right_arm', 'state.left_hand', 'state.right_hand']
action_modality_keys: ['action.left_arm', 'action.right_arm', 'action.left_hand', 'action.right_hand']
Failed to load dataset statistics: 2 validation errors for DatasetStatisticalValues
q01
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
q99
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
Calculating dataset statistics for G1_BlockStacking_Dataset
Collecting all parquet files...:   0%|                                                                                                       | 0/301 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 315, in _get_metadata
    DatasetStatisticalValues.model_validate(stat)
  File "/home/metrum/.local/lib/python3.10/site-packages/pydantic/main.py", line 627, in model_validate
    return cls.__pydantic_validator__.validate_python(
pydantic_core._pydantic_core.ValidationError: 2 validation errors for DatasetStatisticalValues
q01
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing
q99
  Field required [type=missing, input_value={'mean': [-0.387957602739..., 0.008674941025674343]}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.10/v/missing```

### Versions

base

youliangtan, 2025-03-26:
The norm values for `q01` `q99` should be automatically generated in the `stats.json` similar to [this](https://github.com/NVIDIA/Isaac-GR00T/blob/main/demo_data/robot_sim.PickNPlace/meta/stats.json). Could you provide a way to reproduce this error

Rohith-Scalers, 2025-03-27:
Thanks @youliangtan it did generate , the problem was with git lfs , i worked after re-installation

closed 2025-03-27 by Rohith-Scalers