NVIDIA/Isaac-GR00T #58: Corrupted Paraquet files from unitree dataset

Rohith-Scalers, 2025-03-26:
### 🐛 Describe the bug

i am running into corrupted files while fine-tuning for unitree g1 on stacking dataset ? is the current dataset test with gr00t or any changes required to the base dataset ? 
``` ====================================================================================================
========================================= Humanoid Dataset =========================================
====================================================================================================
Traceback (most recent call last):
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 162, in <module>
    load_dataset(args.data_path, args.embodiment_tag, args.video_backend)
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 110, in load_dataset
    resp = dataset[7]
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 488, in __getitem__
    return self.transforms(self.get_step_data(trajectory_id, base_index))
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 518, in get_step_data
    self.curr_traj_data = self.get_trajectory_data(trajectory_id)
  File "/home/metrum/rohith-groot/Isaac-GR00T/gr00t/data/dataset.py", line 535, in get_trajectory_data
    return pd.read_parquet(parquet_path)
  File "/home/metrum/.local/lib/python3.10/site-packages/pandas/io/parquet.py", line 667, in read_parquet
    return impl.read(
  File "/home/metrum/.local/lib/python3.10/site-packages/pandas/io/parquet.py", line 274, in read
    pa_table = self.api.parquet.read_table(
  File "/home/metrum/.local/lib/python3.10/site-packages/pyarrow/parquet/core.py", line 2956, in read_table
    dataset = _ParquetDatasetV2(
  File "/home/metrum/.local/lib/python3.10/site-packages/pyarrow/parquet/core.py", line 2496, in __init__
    [fragment], schema=schema or fragment.physical_schema,
  File "pyarrow/_dataset.pyx", line 1358, in pyarrow._dataset.Fragment.physical_schema.__get__
  File "pyarrow/error.pxi", line 154, in pyarrow.lib.pyarrow_internal_check_status
  File "pyarrow/error.pxi", line 91, in pyarrow.lib.check_status
pyarrow.lib.ArrowInvalid: Could not open Parquet input source '<Buffer>': Parquet magic bytes not found in footer. Either the file is corrupted or this is not a parquet file. ```

### Versions

base

karpenet, 2025-03-26:
Did you install `git lfs` before cloning the dataset repo? If not, it is know to cause this issue.

Installation: https://askubuntu.com/questions/799341/how-do-i-install-git-lfs

Rohith-Scalers, 2025-03-27:
Thanks @karpenet i got it working after re-installation , but still there is this issue , the code is loading specially dataset[7] ? is it due to some hardcoding ? thank you :)

```
~/rohith-groot/Isaac-GR00T$ python3 scripts/load_dataset.py --data_path /home/metrum/rohith-groot/G1_BlockStacking_Dataset/ --embodiment_tag new_embodiment
/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/albumentations/__init__.py:13: UserWarning: A new version of Albumentations is available: 2.0.5 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
("Valid modality_keys for debugging:: {'state': ['state.left_arm', "
 "'state.right_arm', 'state.left_hand', 'state.right_hand'], 'action': "
 "['action.left_arm', 'action.right_arm', 'action.left_hand', "
 "'action.right_hand'], 'video': ['video.cam_right_high', "
 "'video.cam_left_wrist', 'video.cam_right_wrist'], 'annotation': "
 "['annotation.human.task_description']} \n")
state_modality_keys: ['state.left_arm', 'state.right_arm', 'state.left_hand', 'state.right_hand']
action_modality_keys: ['action.left_arm', 'action.right_arm', 'action.left_hand', 'action.right_hand']
Initialized dataset G1_BlockStacking_Dataset with EmbodimentTag.NEW_EMBODIMENT

====================================================================================================
========================================= Humanoid Dataset =========================================
====================================================================================================
failure check
Traceback (most recent call last):
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 164, in <module>
    load_dataset(args.data_path, args.embodiment_tag, args.video_backend)
  File "/home/metrum/rohith-groot/Isaac-GR00T/scripts/load_dataset.py", line 111, in load_dataset
    print(f"dataset[1]: {dataset[1]}")
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/gr00t/data/dataset.py", line 488, in __getitem__
    return self.transforms(self.get_step_data(trajectory_id, base_index))
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/gr00t/data/dataset.py", line 522, in get_step_data
    data[key] = self.get_data_by_modality(trajectory_id, modality, key, base_index)
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/gr00t/data/dataset.py", line 782, in get_data_by_modality
    return self.get_video(trajectory_id, key, base_index)
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/gr00t/data/dataset.py", line 652, in get_video
    return get_frames_by_timestamps(
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/gr00t/utils/video.py", line 64, in get_frames_by_timestamps
    vr = decord.VideoReader(video_path, **video_backend_kwargs)
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/decord/video_reader.py", line 54, in __init__
    self._handle = _CAPI_VideoReaderGetVideoReader(
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/decord/_ffi/_ctypes/function.py", line 173, in __call__
    check_call(_LIB.DECORDFuncCall(
  File "/home/metrum/rohith-groot/.venv/lib/python3.10/site-packages/decord/_ffi/base.py", line 78, in check_call
    raise DECORDError(err_str)
decord._ffi.base.DECORDError: [07:32:44] /github/workspace/src/video/video_reader.cc:151: Check failed: st_nb >= 0 (-1128613112 vs. 0) ERROR cannot find video stream with wanted index: -1```

karpenet, 2025-03-27:
The issue seems to be with `line 111` you added in `load_dataset.py`:
`print(f"dataset[1]: {dataset[1]}")`

To print data keys or vizualize dataset, please refer to this tutorial:
https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/0_load_dataset.ipynb

Rohith-Scalers, 2025-03-27:
Yaa I modified the code a bit , but that was not problem the error was caused due to wrong flash attention package , thanks @karpenet 😀

closed 2025-03-27 by Rohith-Scalers

robotics-qc, 2025-04-07:
Have the same issue on my side too - correct versions are installed

I cloned the g1_stacking dataset and now I am trying to generate the metadata json file  using :
```
python scripts/load_dataset.py --data_path /home/robotics-qc/Documents/Humanoid/Isaac-GR00T/G1_BlockStacking_Dataset --embodiment_tag new_embodiment
```

But the output i get is:
```
UserWarning: A new version of Albumentations is available: 2.0.5 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
("Valid modality_keys for debugging:: {'state': ['state.left_arm', "
 "'state.right_arm', 'state.left_hand', 'state.right_hand'], 'action': "
 "['action.left_arm', 'action.right_arm', 'action.left_hand', "
 "'action.right_hand'], 'video': ['video.cam_right_high', "
 "'video.cam_left_wrist', 'video.cam_right_wrist'], 'annotation': "
 "['annotation.human.task_description']} \n")
state_modality_keys: ['state.left_arm', 'state.right_arm', 'state.left_hand', 'state.right_hand']
action_modality_keys: ['action.left_arm', 'action.right_arm', 'action.left_hand', 'action.right_hand']
Initialized dataset G1_BlockStacking_Dataset with EmbodimentTag.NEW_EMBODIMENT

====================================================================================================
========================================= Humanoid Dataset =========================================
====================================================================================================
Traceback (most recent call last):
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/scripts/load_dataset.py", line 162, in <module>
    load_dataset(args.data_path, args.embodiment_tag, args.video_backend)
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/scripts/load_dataset.py", line 110, in load_dataset
    resp = dataset[7]
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/gr00t/data/dataset.py", line 488, in __getitem__
    return self.transforms(self.get_step_data(trajectory_id, base_index))
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/gr00t/data/dataset.py", line 522, in get_step_data
    data[key] = self.get_data_by_modality(trajectory_id, modality, key, base_index)
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/gr00t/data/dataset.py", line 782, in get_data_by_modality
    return self.get_video(trajectory_id, key, base_index)
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/gr00t/data/dataset.py", line 652, in get_video
    return get_frames_by_timestamps(
  File "/home/robotics-qc/Documents/Humanoid/Isaac-GR00T/gr00t/utils/video.py", line 64, in get_frames_by_timestamps
    vr = decord.VideoReader(video_path, **video_backend_kwargs)
  File "/home/robotics-qc/anaconda3/envs/gr00t/lib/python3.10/site-packages/decord/video_reader.py", line 54, in __init__
    self._handle = _CAPI_VideoReaderGetVideoReader(
  File "/home/robotics-qc/anaconda3/envs/gr00t/lib/python3.10/site-packages/decord/_ffi/_ctypes/function.py", line 173, in __call__
    check_call(_LIB.DECORDFuncCall(
  File "/home/robotics-qc/anaconda3/envs/gr00t/lib/python3.10/site-packages/decord/_ffi/base.py", line 78, in check_call
    raise DECORDError(err_str)
decord._ffi.base.DECORDError: [15:52:32] /github/workspace/src/video/video_reader.cc:151: Check failed: st_nb >= 0 (-1128613112 vs. 0) ERROR cannot find video stream with wanted index: -1
```

robotics-qc, 2025-04-07:
A possible workaround might be to change the backend version to torchvision

```
python scripts/load_dataset.py --data_path /home/robotics-qc/Documents/Humanoid/Isaac-GR00T/G1_BlockStacking_Dataset --embodiment_tag new_embodiment --video_backend torchvision_av
```

youliangtan, 2025-04-07:
Yes by default we uses `decord` since we are using h264 encoding. change to `torchvision` as video backend if using other video encoding.

RZetko, 2025-04-08:
Thanks a lot @robotics-qc @youliangtan. Using --video_backend torchvision_av flag worked like a charm.