NVIDIA/Isaac-GR00T #61: Single arm data missing + Request of GR-1 simulation benchmark data

youthHan, 2025-03-26:
### 🚀 The feature, motivation and pitch

Hi, it's a great effort in open-sourcing the codes and the data. However, I failed to find the single-arm data (single_panda_gripper.*) mentioned in the HF readme (https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim) and the GR-1 tabletop benchmark (which is claimed to be public in the report). Is there a potential schedule for the release of these components? Or maybe I missed them somewhere.
Thank you.

youliangtan, 2025-03-26:
I see a similar issue here: https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/discussions/2

Maybe @avnishn @loicmagne can comment on this.

avnishn, 2025-04-03:
all the data is now uploaded

closed 2025-04-07 by youliangtan

pmh9960, 2025-04-09:
I still can't download the `single_panda_gripper` dataset using the Hugging Face CLI, as described in #99.  
Could you please take a look at my issue?  
Thank you!