NVIDIA/Isaac-GR00T #64: Is it actually following a prompt?

tekntrash, 2025-03-27:
### 🐛 Describe the bug

We tried it with the demo_data/robot_sim.PickNPlace which has tasks such as pick the squash from the counter and place it in the plate. However, if that task is replaced with a meaningless text ('e') the movements created are the same:
         2.24670301e-01]]), 'action.left_hand': array([[-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ]]), 'action.right_hand': array([[-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ],
       [-1.5, -1.5, -1.5, -1.5, -3. ,  3. ]]), 'annotation.human.action.task_description': ['e']}

Thus, it is unclear if it understands what it is supposed to do. Also, being able to replace the LLM with another like deepseek could improve the platform (probably that is already implemented)

### Versions

Python 3.10.12
absl-py==2.2.0
accelerate==1.2.1
action-msgs==1.2.1
action-tutorials-interfaces==0.20.4
action-tutorials-py==0.20.4
actionlib-msgs==4.2.4
aiosignal==1.3.2
albucore==0.0.17
albumentations==1.4.18
ament-cmake-test==1.3.9
ament-copyright==0.12.11
ament-cppcheck==0.12.11
ament-cpplint==0.12.11
ament-flake8==0.12.11
ament-index-python==1.4.0
ament-lint==0.12.11
ament-lint-cmake==0.12.11
ament-package==0.14.0
ament-pep257==0.12.11
ament-uncrustify==0.12.11
ament-xmllint==0.12.11
angles==1.15.0
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
astunparse==1.6.3
attrs==25.3.0
av==12.3.0
blessings==1.7
builtin-interfaces==1.2.1
cachetools==5.5.2
certifi==2025.1.31
charset-normalizer==3.4.1
click==8.1.8
cloudpickle==3.1.1
composition-interfaces==1.2.1
contourpy==1.3.1
cramjam==2.9.1
cv-bridge==3.2.1
cycler==0.12.1
decord==0.6.0
demo-nodes-py==0.20.4
diagnostic-msgs==4.2.4
diffusers==0.30.2
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.16
domain-coordinator==0.10.0
einops==0.8.1
eval_type_backport==0.2.2
example-interfaces==0.9.3
examples-rclpy-executors==0.15.1
examples-rclpy-minimal-action-client==0.15.1
examples-rclpy-minimal-action-server==0.15.1
examples-rclpy-minimal-client==0.15.1
examples-rclpy-minimal-publisher==0.15.1
examples-rclpy-minimal-service==0.15.1
examples-rclpy-minimal-subscriber==0.15.1
exceptiongroup==1.2.2
Farama-Notifications==0.0.4
fastparquet==2024.11.0
filelock==3.18.0
flash-attn==2.7.1.post4
flatbuffers==25.2.10
fonttools==4.56.0
frozenlist==1.5.0
fsspec==2025.3.0
fvcore==0.1.5.post20221221
gast==0.6.0
geometry-msgs==4.2.4
gitdb==4.0.12
GitPython==3.1.44
google-auth==2.38.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
-e git+https://github.com/NVIDIA/Isaac-GR00T@153ee6fee23d4c3bf52cb4b520ddd258f98d017b#egg=gr00t
grpcio==1.71.0
gymnasium==1.0.0
h5py==3.12.1
hf_transfer==0.1.9
huggingface-hub==0.29.3
hydra-core==1.3.2
idna==3.10
image-geometry==3.2.1
imageio==2.34.2
importlib_metadata==8.6.1
iniconfig==2.1.0
interactive-markers==2.3.2
iopath==0.1.10
Jinja2==3.1.6
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
keras==2.15.0
kiwisolver==1.4.8
kornia==0.7.4
kornia_rs==0.1.8
laser-geometry==2.4.0
launch==1.0.6
launch-ros==0.19.7
launch-testing==1.0.6
launch-testing-ros==0.19.7
launch-xml==1.0.6
launch-yaml==1.0.6
lazy_loader==0.4
libclang==18.1.1
lifecycle-msgs==1.2.1
llvmlite==0.44.0
logging-demo==0.20.4
map-msgs==2.1.0
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
mdurl==0.1.2
message-filters==4.3.4
ml-dtypes==0.2.0
mpmath==1.3.0
msgpack==1.1.0
nav-msgs==4.2.4
networkx==3.4.2
numba==0.61.0
numpy==1.26.4
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.2.2
omegaconf==2.3.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
opt_einsum==3.4.0
osrf-pycommon==2.0.2
packaging==24.2
pandas==2.2.3
pcl-msgs==1.0.0
peft==0.14.0
pendulum-msgs==0.20.4
pettingzoo==1.24.3
pillow==11.1.0
pipablepytorch3d==0.7.6
platformdirs==4.3.7
pluggy==1.5.0
portalocker==3.1.1
protobuf==3.20.3
psutil==7.0.0
pyarrow==14.0.1
pyasn1==0.6.1
pyasn1_modules==0.4.1
pyav==14.2.1
pydantic==2.10.6
pydantic_core==2.27.2
Pygments==2.19.1
pyparsing==3.2.3
pytest==8.3.5
python-dateutil==2.9.0.post0
python-qt-binding==1.1.2
pytz==2025.2
PyYAML==6.0.2
pyzmq==26.3.0
qt-dotgraph==2.2.3
qt-gui==2.2.3
qt-gui-cpp==2.2.3
qt-gui-py-common==2.2.3
quality-of-service-demo-py==0.20.4
ray==2.40.0
rcl-interfaces==1.2.1
rclpy==3.3.13
rcutils==5.1.6
referencing==0.36.2
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
resource-retriever==3.1.2
rich==13.9.4
rmw-dds-common==1.6.0
ros2action==0.18.10
ros2bag==0.15.11
ros2cli==0.18.10
ros2component==0.18.10
ros2doctor==0.18.10
ros2interface==0.18.10
ros2launch==0.19.7
ros2lifecycle==0.18.10
ros2multicast==0.18.10
ros2node==0.18.10
ros2param==0.18.10
ros2pkg==0.18.10
ros2run==0.18.10
ros2service==0.18.10
ros2topic==0.18.10
rosbag2-interfaces==0.15.11
rosbag2-py==0.15.11
rosgraph-msgs==1.2.1
rosidl-adapter==3.1.5
rosidl-cli==3.1.5
rosidl-cmake==3.1.5
rosidl-generator-c==3.1.5
rosidl-generator-cpp==3.1.5
rosidl-generator-py==0.14.4
rosidl-parser==3.1.5
rosidl-runtime-py==0.9.3
rosidl-typesupport-c==2.0.1
rosidl-typesupport-cpp==2.0.1
rosidl-typesupport-fastrtps-c==2.2.2
rosidl-typesupport-fastrtps-cpp==2.2.2
rosidl-typesupport-introspection-c==3.1.5
rosidl-typesupport-introspection-cpp==3.1.5
rpds-py==0.23.1
rpyutils==0.2.1
rqt-action==2.0.1
rqt-bag==1.1.5
rqt-bag-plugins==1.1.5
rqt-console==2.0.3
rqt-graph==1.3.1
rqt-gui==1.1.7
rqt-gui-py==1.1.7
rqt-msg==1.2.0
rqt-plot==1.1.2
rqt-publisher==1.5.0
rqt-py-common==1.1.7
rqt-py-console==1.0.2
rqt-reconfigure==1.1.2
rqt-service-caller==1.0.5
rqt-shell==1.0.2
rqt-srv==1.0.3
rqt-topic==1.5.0
rsa==4.9
safetensors==0.5.3
scikit-image==0.25.2
scipy==1.15.2
sensor-msgs==4.2.4
sensor-msgs-py==4.2.4
sentry-sdk==2.24.1
setproctitle==1.3.5
shape-msgs==4.2.4
shtab==1.7.1
simsimd==6.2.1
six==1.17.0
smmap==5.0.2
sros2==0.10.5
statistics-msgs==1.2.1
std-msgs==4.2.4
std-srvs==4.2.4
stereo-msgs==4.2.4
stringzilla==3.12.3
sympy==1.13.1
tabulate==0.9.0
teleop-twist-keyboard==2.4.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.5.0
tf2-geometry-msgs==0.25.7
tf2-kdl==0.25.7
tf2-msgs==0.25.7
tf2-py==0.25.7
tf2-ros-py==0.25.7
tf2-tools==0.25.7
tianshou==0.5.1
tifffile==2025.3.13
timm==1.0.14
tokenizers==0.20.3
tomli==2.2.1
topic-monitor==0.20.4
torch==2.5.1
torchvision==0.20.1
tqdm==4.67.1
trajectory-msgs==4.2.4
transformers==4.45.2
triton==3.1.0
turtlesim==1.4.2
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
unique-identifier-msgs==2.2.1
urllib3==2.3.0
visualization-msgs==4.2.4
wandb==0.18.0
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp==3.21.0
zmq==0.0.0

youliangtan, 2025-03-28:
In general when the prompt is not valid. it will predict the next highly plausible trajectory. In this case, it is still the pick and place task. Open-ended language following is still an active research topic =)

tekntrash, 2025-03-28:
Perhaps the system could just do nothing if the prompt is not valid? Otherwise we end up with a robot that can do unexplainable moves? Dunno, just a thought, hope it makes sense

youliangtan, 2025-04-07:
@tekntrash This can be done on your own dataset too. provide idle behavior if prompt is empty, etc.

closed 2025-04-07 by youliangtan