NVIDIA/Isaac-GR00T #65: hi,if I want to use groot to test on LIBERO and  Bridge V2 Dataset ,what should I do ,I am new learner

Luo-Zhongwei, 2025-03-27:


Luo-Zhongwei, 2025-03-29:
> For any dataset to be compatible with Gr00t, you need to convert it to LeRobot compatible schema with additional metadata files as specified here: https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/LeRobot_compatible_data_schema.md

hi,I read them and find lerobot contains such meta files
```python
├── meta
│   ├── episodes.jsonl
│   ├── episodes_stats.jsonl
│   ├── info.json
│   └── tasks.jsonl
```

and your  groot meta is 
```python
meta
├── episodes.jsonl
├── info.json
├── modality.json
├── stats.json
└── tasks.jsonl
```

is the stats.json equal to episodes_stats.jsonl ?
and I need to add modality.json?

Luo-Zhongwei, 2025-03-29:
> 对于与 Gr00t 兼容的任何数据集，您需要将其转换为 LeRobot 兼容模式，并使用此处指定的其他元数据文件： https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/LeRobot_compatible_data_schema.md

and what is more, for other lerobot datast, what I should do is add a  modality to  file? add this could be ok(from your example)?
```python
{
    "state": {
        "left_arm": {
            "start": 0,
            "end": 7
        },
        "left_hand": {
            "start": 7,
            "end": 13
        },
        "left_leg": {
            "start": 13,
            "end": 19
        },
        "neck": {
            "start": 19,
            "end": 22
        },
        "right_arm": {
            "start": 22,
            "end": 29
        },
        "right_hand": {
            "start": 29,
            "end": 35
        },
        "right_leg": {
            "start": 35,
            "end": 41
        },
        "waist": {
            "start": 41,
            "end": 44
        }
    },
    "action": {
        "left_arm": {
            "start": 0,
            "end": 7
        },
        "left_hand": {
            "start": 7,
            "end": 13
        },
        "left_leg": {
            "start": 13,
            "end": 19
        },
        "neck": {
            "start": 19,
            "end": 22
        },
        "right_arm": {
            "start": 22,
            "end": 29
        },
        "right_hand": {
            "start": 29,
            "end": 35
        },
        "right_leg": {
            "start": 35,
            "end": 41
        },
        "waist": {
            "start": 41,
            "end": 44
        }
    },
    "video": {
        "ego_view": {
            "original_key": "observation.images.ego_view"
        }
    },
    "annotation": {
        "human.action.task_description": {},
        "human.validity": {}
    }
}

```

karpenet, 2025-03-29:
- `stats.json` and `episodes_stats.json` are same.
- Yes, you only need to add `modality.json` if you're finetuning on a supported embodiment. If you want to finetune for a new embodiment, the refer to #67.
- There are some example [modality](getting_started/examples) files here to help you understand better.

I found a [script](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/convert_libero_data_to_lerobot.py) that converts Libero to LeRobot for you!

Luo-Zhongwei, 2025-03-31:
> * `stats.json` and `episodes_stats.json` are same.
> * Yes, you only need to add `modality.json` if you're finetuning on a supported embodiment. If you want to finetune for a new embodiment, the refer to [Using state-action projector for non-GR1 robots in EmbodimentTag #67](https://github.com/NVIDIA/Isaac-GR00T/issues/67).
> * There are some example [modality](getting_started/examples) files here to help you understand better.
> 
> I found a [script](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/convert_libero_data_to_lerobot.py) that converts Libero to LeRobot for you!

is that mean, for all lerobot  dataset, what I should do is to add a modality,json like 

should I change the content of the json or just keep it unchange for all lerobot dataset?

```python
{
    "state": {
        "left_arm": {
            "start": 0,
            "end": 7
        },
        "left_hand": {
            "start": 7,
            "end": 13
        },
        "left_leg": {
            "start": 13,
            "end": 19
        },
        "neck": {
            "start": 19,
            "end": 22
        },
        "right_arm": {
            "start": 22,
            "end": 29
        },
        "right_hand": {
            "start": 29,
            "end": 35
        },
        "right_leg": {
            "start": 35,
            "end": 41
        },
        "waist": {
            "start": 41,
            "end": 44
        }
    },
    "action": {
        "left_arm": {
            "start": 0,
            "end": 7
        },
        "left_hand": {
            "start": 7,
            "end": 13
        },
        "left_leg": {
            "start": 13,
            "end": 19
        },
        "neck": {
            "start": 19,
            "end": 22
        },
        "right_arm": {
            "start": 22,
            "end": 29
        },
        "right_hand": {
            "start": 29,
            "end": 35
        },
        "right_leg": {
            "start": 35,
            "end": 41
        },
        "waist": {
            "start": 41,
            "end": 44
        }
    },
    "video": {
        "ego_view": {
            "original_key": "observation.images.ego_view"
        }
    },
    "annotation": {
        "human.action.task_description": {},
        "human.validity": {}
    }
}
```

Luo-Zhongwei, 2025-03-31:
> > * `stats.json` and `episodes_stats.json` are same.
> > * Yes, you only need to add `modality.json` if you're finetuning on a supported embodiment. If you want to finetune for a new embodiment, the refer to [Using state-action projector for non-GR1 robots in EmbodimentTag #67](https://github.com/NVIDIA/Isaac-GR00T/issues/67).
> > * There are some example [modality](getting_started/examples) files here to help you understand better.
> > 
> > I found a [script](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/convert_libero_data_to_lerobot.py) that converts Libero to LeRobot for you!
> 
> is that mean, for all lerobot dataset, what I should do is to add a modality,json like
> 
> should I change the content of the json or just keep it unchange for all lerobot dataset?
> 
> {
>     "state": {
>         "left_arm": {
>             "start": 0,
>             "end": 7
>         },
>         "left_hand": {
>             "start": 7,
>             "end": 13
>         },
>         "left_leg": {
>             "start": 13,
>             "end": 19
>         },
>         "neck": {
>             "start": 19,
>             "end": 22
>         },
>         "right_arm": {
>             "start": 22,
>             "end": 29
>         },
>         "right_hand": {
>             "start": 29,
>             "end": 35
>         },
>         "right_leg": {
>             "start": 35,
>             "end": 41
>         },
>         "waist": {
>             "start": 41,
>             "end": 44
>         }
>     },
>     "action": {
>         "left_arm": {
>             "start": 0,
>             "end": 7
>         },
>         "left_hand": {
>             "start": 7,
>             "end": 13
>         },
>         "left_leg": {
>             "start": 13,
>             "end": 19
>         },
>         "neck": {
>             "start": 19,
>             "end": 22
>         },
>         "right_arm": {
>             "start": 22,
>             "end": 29
>         },
>         "right_hand": {
>             "start": 29,
>             "end": 35
>         },
>         "right_leg": {
>             "start": 35,
>             "end": 41
>         },
>         "waist": {
>             "start": 41,
>             "end": 44
>         }
>     },
>     "video": {
>         "ego_view": {
>             "original_key": "observation.images.ego_view"
>         }
>     },
>     "annotation": {
>         "human.action.task_description": {},
>         "human.validity": {}
>     }
> }

by the way, [modality](https://github.com/NVIDIA/Isaac-GR00T/issues/getting_started/examples)  is  404

bailvwangzi, 2025-04-01:
> * `stats.json` and `episodes_stats.json` are same.
> * Yes, you only need to add `modality.json` if you're finetuning on a supported embodiment. If you want to finetune for a new embodiment, the refer [...]

this script can't generate videos, how to get LIBERO videos?

Catchip, 2025-04-07:
You can find Libero in lerobot format here [OpenX-LeRobot](https://huggingface.co/collections/IPEC-COMMUNITY/openx-lerobot-67c29b2ee5911f17dbea635e)

bailvwangzi, 2025-04-14:
I have finetuned groot on LIBERO, the result is not good enough, Libero-Goal success rate=71%. Maybe i have some wrong setting? Do you have plan to give more sim results to compare with other method? @karpenet

Catchip, 2025-04-17:
@bailvwangzi  Hi, could you please share the modality.json used by LIBERO?

Jerryisqx, 2025-04-17:
@bailvwangzi Hi, How do you integrate gr00T with the LIBERO simulation? Did you adjust from pi0's libero sim script?

bailvwangzi, 2025-04-18:
> modality

{ "state": { "state": { "start": 0, "end": 8 }}, "action": { "actions": { "start": 0, "end": 7 }}, "video": { "image": { "original_key": "observation.images.image" } , "wrist_image": { "original_key": "observation.images.wrist_image" }}, "annotation": { "human.task_description": { "original_key": "task_index" } } }

Catchip, 2025-04-27:
@Jerryisqx I adjust from pi0's libero sim script, and evaluate successfully.

VV940, 2025-05-16:
> 您可以在此处找到 lerobot 格式的 Libero [OpenX-LeRobot](https://huggingface.co/collections/IPEC-COMMUNITY/openx-lerobot-67c29b2ee5911f17dbea635e)

decord._ffi.base.DECORDError: [14:32:47] /github/workspace/src/video/video_reader.cc:151: Check failed: st_nb >= 0 (-1128613112 vs. 0) ERROR cannot find video stream with wanted index: -1；
Hello, when using the data set of this link to read videos, some videos will report errors due to wave distortion. Could you please tell me how to handle this situation?

Catchip, 2025-05-16:
@VV940 use the torchvision_av backend

VV940, 2025-05-16:
> 

> [@VV940](https://github.com/VV940) use the torchvision_av video backend
> 
> python scripts/gr00t_finetune.py --video-backend torchvision_av --dataset-path "/path/to/dataset" --num-gpus NUM_GPU --max_steps NUM_STEP --output_dir "path/to/output" --data_config DATA_CONFIG

It works.Thank you!

VV940, 2025-05-16:
> 我在 LIBERO 上微调了 groot，结果不够好，Libero-Goal 成功率=71%。也许我设置错误？您是否打算提供更多的 sim 结果以与其他方法进行比较？
Could you please explain how the success rate of 71% was determined? I ran the evaluation code, and it only showed the MSE error for a single trajectory. May I ask if you conducted real-device experiments?

Catchip, 2025-05-21:
@VV940 You can refer to this [issue](https://github.com/NVIDIA/Isaac-GR00T/issues/136#issue-3020264554)

Catchip, 2025-05-26:
@VV940 
The meanings of "action space" and "state space" should be analyzed based on the specific dataset you are using. For example, if you are using Libero dataset from [OpenX-LeRobot](https://huggingface.co/collections/IPEC-COMMUNITY/openx-lerobot-67c29b2ee5911f17dbea635e), you can find that it was generated by this [script](https://github.com/Tavish9/any4lerobot/blob/main/openx2lerobot/openx_rlds.py), and you may discover the following part in the code:
```python
def generate_features_from_raw(builder: tfds.core.DatasetBuilder, use_videos: bool = True):
    dataset_name = Path(builder.data_dir).parent.name

    state_names = [f"motor_{i}" for i in range(8)]
    if dataset_name in OXE_DATASET_CONFIGS:
        state_encoding = OXE_DATASET_CONFIGS[dataset_name]["state_encoding"]
        if state_encoding == StateEncoding.POS_EULER:
            state_names = ["x", "y", "z", "roll", "pitch", "yaw", "pad", "gripper"]
            if "libero" in dataset_name:
                state_names = ["x", "y", "z", "roll", "pitch", "yaw", "gripper", "gripper"]  # 2D gripper state
        elif state_encoding == StateEncoding.POS_QUAT:
            state_names = ["x", "y", "z", "rx", "ry", "rz", "rw", "gripper"]
        elif state_encoding == StateEncoding.JOINT:
            state_names = [f"motor_{i}" for i in range(7)] + ["gripper"]
            state_obs_keys = OXE_DATASET_CONFIGS[dataset_name]["state_obs_keys"]
            pad_count = state_obs_keys[:-1].count(None)
            state_names[-pad_count - 1 : -1] = ["pad"] * pad_count
            state_names[-1] = "pad" if state_obs_keys[-1] is None else state_names[-1]

    action_names = [f"motor_{i}" for i in range(8)]
    if dataset_name in OXE_DATASET_CONFIGS:
        action_encoding = OXE_DATASET_CONFIGS[dataset_name]["action_encoding"]
        if action_encoding == ActionEncoding.EEF_POS:
            action_names = ["x", "y", "z", "roll", "pitch", "yaw", "gripper"]
        elif action_encoding == ActionEncoding.JOINT_POS:
            action_names = [f"motor_{i}" for i in range(7)] + ["gripper"]
```

You can infer from this code snippet that the state space corresponds to ```["x", "y", "z", "roll", "pitch", "yaw", "gripper", "gripper"] ```, and the action space corresponds to  ```["x", "y", "z", "roll", "pitch", "yaw", "gripper"]```. In this example, since the two grippers are symmetrical, the action can be controlled using just one dimension for the gripper.

However, the state space does not always have one more dimension than the action space, as shown in this [example](https://github.com/OpenDriveLab/AgiBot-World/issues/35). It still requires specific analysis.