NVIDIA/Isaac-GR00T #68: Fine-tuning with xarm6 + gripper on custom dataset.

pratikgehlott, 2025-03-27:
Hi Team,

Thank you for your excellent work!

I have a few questions about the setup.

I collected 180 episodes using the xArm6 with a gripper and uploaded them to the hub - [Hugging face link](https://huggingface.co/datasets/nimitvasavat/Gr00t_lerobot)

Dataset Information:

1. **State: 8 values (absolute values): x, y, z, qx, qy, qz, qw, gripper**
2. **Action: 7 values (delta values from previous action): x, y, z, roll, pitch, yaw, gripper**
3. **video.image: logitech camera (3rd person view)**
4. **video.back_image: realsense camera placed behind the robotic arm**
5. **video.wrist_image: realsense camera on the wrist**

For fine-tuning, I used the following code - 

**data_config.py**

```
class xArm6DataConfig(BaseDataConfig):
    video_keys = ["video.image", "video.back_image", "video.wrist_image"]
    state_keys = ["state.arm", "state.gripper"]
    action_keys = ["action.arm", "action.gripper"]
    language_keys = ["annotation.human.task_description"]
    observation_indices = [0]
    action_indices = list(range(16))

    def modality_config(self) -> dict[str, ModalityConfig]:
        video_modality = ModalityConfig(
            delta_indices=self.observation_indices,
            modality_keys=self.video_keys,
        )

        state_modality = ModalityConfig(
            delta_indices=self.observation_indices,
            modality_keys=self.state_keys,
        )

        action_modality = ModalityConfig(
            delta_indices=self.action_indices,
            modality_keys=self.action_keys,
        )

        language_modality = ModalityConfig(
            delta_indices=self.observation_indices,
            modality_keys=self.language_keys,
        )

        modality_configs = {
            "video": video_modality,
            "state": state_modality,
            "action": action_modality,
            "language": language_modality,
        }

        return modality_configs
    
    def transform(self) -> ModalityTransform:
        transforms = [
            # video transforms
            VideoToTensor(apply_to=self.video_keys),
            VideoCrop(apply_to=self.video_keys, scale=0.95),
            VideoResize(apply_to=self.video_keys, height=224, width=224, interpolation="linear"),
            VideoColorJitter(
                apply_to=self.video_keys,
                brightness=0.3,
                contrast=0.4,
                saturation=0.5,
                hue=0.08,
            ),
            VideoToNumpy(apply_to=self.video_keys),
            # state transforms
            StateActionToTensor(apply_to=self.state_keys),
            StateActionTransform(
                apply_to=self.state_keys,
                normalization_modes={key: "min_max" for key in self.state_keys},
            ),
            # action transforms
            StateActionToTensor(apply_to=self.action_keys),
            StateActionTransform(
                apply_to=self.action_keys,
                normalization_modes={key: "min_max" for key in self.action_keys},
            ),
            # concat transforms
            ConcatTransform(
                video_concat_order=self.video_keys,
                state_concat_order=self.state_keys,
                action_concat_order=self.action_keys,
            ),
            # model-specific transform
            GR00TTransform(
                state_horizon=len(self.observation_indices),
                action_horizon=len(self.action_indices),
                max_state_dim=64,
                max_action_dim=32,
            ),
        ]
        return ComposedModalityTransform(transforms=transforms)
```

Command to fine tune - 
`
python scripts/gr00t_finetune.py    --dataset-path getting_started/vera_data/Gr00t_lerobot/    --num-gpus 1    --output-dir ~/vera-checkpoints_second     --max-steps 20000    --data-config xarm6    --video-backend torchvision_av `

I have a question: when I ran this model using the policy inference code, the xArm6 did not perform accurately. We are using the action output from the model as a set of delta changes in postition, i.e., if our initial position is x, y, z, roll, pitch, yaw then the first 6 values of each action represents the "delta" added to the aforementioned coordinate values. The subsequent action deltas are then added to the updated position till we complete the set of actions. For example, if at the start we have

`x = 0.12 y = 0.14 z = 1.2 roll = 0.01 pitch = 0.23 yaw = 0.145`

and the first action is `[0.03, 0.065, 0.02, 0.003, 0.024, 0.01, 1]` then we move our robot arm to the following position:

`[0.15, 0.205, 1.22, 0.013, 0.254, 0.155]`, i.e., we assume each action represents the change from current position and add it to get the new position for the next action

Is this setup correct, or do I need to do something else?

**Couple of more questions -** 

1. After running the inference code, I get an actions shape of (16, 6) and (16, 1). Does this mean that the model is set up to perform 16 steps incrementally, or some other transformation needs to be applied before sending the actions to the robot?
2. Does the robotic arm require calibration? While testing the policy, I keep the robot in the same position as it was at the start of the single-episode data collection.

Thank you

youliangtan, 2025-03-30:
1. Quaternion as input to the NN is generally not recommended ([ref](https://arxiv.org/abs/1812.07035)). you can use the [RotationTransform](https://github.com/NVIDIA/Isaac-GR00T/blob/0829414de5b28df8160be6b732972f12a2c04625/gr00t/data/transform/state_action.py#L29) to convert to a "rotation6d" format for the state input.
2. I don't get why you are getting `(16, 6) and (16, 1)`. In general the default 16 values is the size of the action chunk. (16 steps in the future)
3. The calibration step is crucial to ensure consistent "setup" during data collection and evaluation.
4. By quickly glancing your provided dataset above, I'm curious why the state and action are not being plotted out on the lerobot visualizer here: https://huggingface.co/spaces/lerobot/visualize_dataset?dataset=nimitvasavat%2FGr00t_lerobot&episode=0

pratikgehlott, 2025-03-31:
Hi @youliangtan , thank you for your response.

2. I obtain outputs with shapes (16, 6) and (16, 1). I interpret the (16, 6) shape as representing 16 future steps with 6 values each—specifically, the deltas Δ(x, y, z, roll, pitch, yaw)—and the (16, 1) shape as representing the state of the gripper. My concern is whether I should treat the output as delta changes or if there is another interpretation. We think the output is in this format as OpenVLA uses a similar format. Can you please help me figure this out, as well as let me know if the output is instead joint angles, absolute or relative?

3. Yes, I keep my setup consistent for both episode collection and inference.

closed 2025-04-07 by pratikgehlott