NVIDIA/Isaac-GR00T #69: TypeError: Object of type function is not JSON serializable The format kwargs must be JSON serializable, but key 'transform' isn't.

Luo-Zhongwei, 2025-03-28:
### 🐛 Describe the bug

hi.I write a code to download hf lerobot dataset ,but get some error using lerobot dataset, 

error:

```python
Traceback (most recent call last):███████████████████████████████████████| 35.7M/35.7M [00:30<00:00, 925kB/s]
  File "/root/autodl-tmp/lerobot/examples/my_download.py", line 36, in <module>
    ds.hf_dataset.save_to_disk(dataset_path)
  File "/root/miniconda3/envs/groot/lib/python3.10/site-packages/datasets/arrow_dataset.py", line 1518, in save_to_disk
    raise TypeError(
TypeError: Object of type function is not JSON serializable
The format kwargs must be JSON serializable, but key 'transform' isn't.
(groot) (base) root@autodl-container-6d7f4f8bb4-1352f3de:~/autodl-tmp/lerobot# 
```

my code:
```python
import os
import shutil
from huggingface_hub import HfApi
from lerobot.common.datasets.lerobot_dataset import LeRobotDataset
import os
os.environ["HF_ENDPOINT"] = "https://hf-mirror.com"  # Hugging Face的国内镜像
# 设置本地缓存目录
cache_dir = "/root/autodl-tmp/lerobot_data/"
os.makedirs(cache_dir, exist_ok=True)

hub_api = HfApi()
repo_infos = hub_api.list_datasets(task_categories="robotics", tags=["LeRobot"])

total_size = 0
dataset_count = 0

# 默认缓存路径
hf_cache_dir = os.path.expanduser("~/.cache/huggingface/datasets")

# 下载最多三个数据集到本地缓存目录，并显示进度条
for info in repo_infos:
    if dataset_count >= 4:
        print("已下载三个数据集，终止程序。")
        break

    repo_id = info.id
    dataset_path = os.path.join(cache_dir, repo_id.replace("/", "_"))

    # 检查是否已经下载
    if not os.path.exists(dataset_path) or not os.listdir(dataset_path):
        print(f"开始下载数据集: {repo_id}")
        ds = LeRobotDataset(repo_id)
        # 使用内置进度条显示下载进度，下载到默认缓存目录
        
        # 保存数据集到指定的路径
        ds.hf_dataset.save_to_disk(dataset_path)

        # 删除默认缓存中的数据集
        cached_repo_dir = os.path.join(hf_cache_dir, repo_id.replace("/", "___"))
        if os.path.exists(cached_repo_dir):
            shutil.rmtree(cached_repo_dir)

        dataset_size = sum(os.path.getsize(os.path.join(root, file)) for root, _, files in os.walk(dataset_path) for file in files)
        total_size += dataset_size
        dataset_count += 1
        avg_size_gb = (total_size / dataset_count) / (1024**3)

        print(f"数据集下载完成: {repo_id}，当前数据集大小: {dataset_size / (1024**3):.2f} GB，平均大小: {avg_size_gb:.2f} GB")

        if total_size > (200 * 1024**3 - avg_size_gb * 1024**3):
            print("⚠️ 数据集总大小超过限制 (200GB - 平均数据集大小)，终止下载程序！")
[... 250 lines cut ...]
torchaudio                   2.5.0
torchvision                  0.20.1+cu124
tqdm                         4.67.1
traitlets                    5.14.3
transformers                 4.47.0
transforms3d                 0.4.2
trimesh                      4.6.4
triton                       3.1.0
truststore                   0.10.1
typeguard                    2.13.3
typing_extensions            4.12.2
tzdata                       2025.1
tzlocal                      5.3.1
urllib3                      2.3.0
virtualenv                   20.29.3
wandb                        0.19.8
wcwidth                      0.2.13
Werkzeug                     3.1.3
wheel                        0.45.1
wrapt                        1.14.1
Yapsy                        1.12.2
zipp                         3.21.0
zstandard                    0.23.0

```

pratikgehlott, 2025-03-28:
Just use the below lines to download the dataset-

`pip install -U "huggingface_hub[cli]"`

`
huggingface-cli download --repo-type dataset youliangtan/so100_strawberry_grape --local-dir ./demo_data/so100_strawberry_grape `

karpenet, 2025-03-28:
Install `git lfs` before you download using cli.
https://askubuntu.com/questions/799341/how-do-i-install-git-lfs

closed 2025-03-28 by Luo-Zhongwei