NVIDIA/Isaac-GR00T #70: Output action is delta action or desired state?

SteveOUO, 2025-03-28:
Hello, thank you for your great work and for open-sourcing this project.
I ran your example 3_new_embodiment_finetuning and generated the following plot. In the plot, the curve of the state joints is very similar to the other two curves. Does this mean that the output action represents the desired state instead of the delta action?
[image]

youliangtan, 2025-03-28:
In absolute position control, the state curve is essentially the current state and the action is the "target state" which the robot wish to reach. Thus the offset "delay" of the state to reach the target action

closed 2025-03-28 by youliangtan

youliangtan, 2025-04-02:
Yes you are right because of the action chunk this the trajectory of every action chunk. X-axis is time