NVIDIA/Isaac-GR00T #74: Error during inference

Jerryisqx, 2025-03-30:
### 🐛 Describe the bug

Hi, I'm trying to deploy the gr00t on cluster and inference, here is the error I meet:

Environment configuration:
```
CONDA_OVERRIDE_CUDA=11.8 conda create --name gr00t python=3.10 pytorch torchvision pytorch-cuda=11.8 --channel pytorch --channel nvidia
conda activate gr00t

export PATH=/PATH To my cuda 11.8/:${PATH}
export LD_LIBRARY_PATH=/PATH To my cuda 11.8/lib64

pip install --upgrade setuptools
pip install -e .
pip install --no-build-isolation flash-atten=2.6.1 --use-pep517
```
I use cuda11.8 because the cluster currently can not support cuda12, and thus I'm trying to install the relative flash attention 2.6.1 to match the cuda version.

Then, I put the inference notebook in the getting started folder into one script file `inference.py` and run it, got the error:
```
python /path to inference.py/inference.py

2025-03-30 15:19:44.394653: I external/local_tsl/tsl/cuda/cudart_stub.cc:31] Could not find cuda drivers on your machine, GPU will not be used.
2025-03-30 15:19:44.440621: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-03-30 15:19:44.440691: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-03-30 15:19:44.441933: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-03-30 15:19:44.448598: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-03-30 15:19:45.867294: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT

Fetching 6 files:   0%|          | 0/6 [00:00<?, ?it/s]
Fetching 6 files: 100%|██████████| 6/6 [00:00<00:00, 12075.73it/s]
/scratch_net/biwidl313_second/Isaac-GR00T/gr00t/model/action_head/cross_attention_dit.py:249: FutureWarning: Accessing config attribute `output_dim` directly via 'DiT' object attribute is deprecated. Please access 'output_dim' over 'DiT's config object instead, e.g. 'unet.config.output_dim'.
  self.proj_out_2 = nn.Linear(self.inner_dim, self.output_dim)
Some weights of the model checkpoint at /scratch_net/biwidl313_second/huggingface_cache/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb were not used when initializing GR00T_N1: ['action_head.decode_layer.bias', 'action_head.decode_layer.weight']
- This IS expected if you are initializing GR00T_N1 from the checkpoint of a model trained on another task or with another architecture (e.g. initializing a BertForSequenceClassification model from a BertForPreTraining model).
- This IS NOT expected if you are initializing GR00T_N1 from the checkpoint of a model that you expect to be exactly identical (initializing a BertForSequenceClassification model from a BertForSequenceClassification model).
Loading pretrained dual brain from /scratch_net/biwidl313_second/huggingface_cache/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /scratch_net/biwidl313_second/huggingface_cache/hub/models--nvidia--GR00T-N1-2B/snapshots/32e1fd2507f7739fad443e6b449c8188e0e02fcb
Total number of DiT parameters:  537803776
Tune action head projector: True
Tune action head diffusion model: True
Tune action head projector: True
Tune action head diffusion model: True
Traceback (most recent call last):
  File "/scratch_net/biwidl313_second/Isaac-GR00T/getting_started/inference.py", line 25, in <module>
    policy = Gr00tPolicy(
  File "/scratch_net/biwidl313_second/Isaac-GR00T/gr00t/model/policy.py", line 108, in __init__
    self._load_model(model_path)
  File "/scratch_net/biwidl313_second/Isaac-GR00T/gr00t/model/policy.py", line 236, in _load_model
    model.to(device=self.device)  # type: ignore
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/transformers/modeling_utils.py", line 2958, in to
    return super().to(*args, **kwargs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1340, in to
    return self._apply(convert)
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 900, in _apply
    module._apply(fn)
  [Previous line repeated 3 more times]
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 927, in _apply
    param_applied = fn(param)
  File "/scratch_net/biwidl313/chenqing/conda_envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1326, in convert
    return t.to(
[... 146 lines cut ...]
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.5.0
tianshou==0.5.1
tifffile==2025.3.30
timm==1.0.14
tokenizers==0.20.3
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tqdm==4.67.1
transformers==4.45.2
triton==3.1.0
typeguard==4.4.2
typing_extensions @ file:///croot/typing_extensions_1734714854207/work
tyro==0.9.17
tzdata==2025.2
urllib3 @ file:///croot/urllib3_1737133630106/work
wandb==0.18.0
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp==3.21.0
zmq==0.0.0

youliangtan, 2025-04-18:
We havnt tested on other versions of cuda, but if you manage to get it work, a PR to update the instructions or cuda version would be helpful. Thanks!

AndersonYu7, 2025-08-02:
Hi, I successfully ran both finetuning and inference using the following setup:
- CUDA: 11.8  
- Python: 3.10

- torch==2.7.1+cu118  
- torchvision==0.22.1+cu118  
- torchaudio==2.7.1+cu118  
- torchcodec==0.5  
- flash-attn==2.8.2  

Both fine-tuning and inference ran successfully using this setup.  
Hope this helps

xwjiang2010, 2025-09-06:
@Jerryisqx
Please see above for @AndersonYu7 's input on using CUDA 11.8 (thank you @AndersonYu7 !).

closed 2025-09-06 by xwjiang2010