NVIDIA/Isaac-GR00T #78: Question: Does the GR00T-N1-2B model support action inputs larger than 32 dimensions?

maxdmx, 2025-03-31:
### 📚 The doc issue

Dear NVIDIA GR00T team,

I hope this message finds you well.

I am currently working on fine-tuning the gr00t-n1-2B model, which I downloaded from Hugging Face, for deployment on our robotic platform. However, we've encountered a challenge, as our robot's action space dimension exceeds the model's default maximum of 32 dimensions.

Therefore, I would be most grateful if you could provide some guidance on the following questions:

Could you please confirm whether the gr00t-n1-2B model supports action input dimensions greater than 32?
If it does, would you be so kind as to outline the necessary modifications to accommodate higher action dimensions?
Are there any example code snippets or documentation that you could kindly share, which demonstrate how to modify the model for increased action dimensionality?

I would greatly appreciate any detailed instructions on how to adjust the model to support higher-dimensional action inputs, including any required code changes and configuration adjustments.

HuBocheng, 2025-04-14:
The same question

HuFY-dev, 2025-06-11:
You can try to create your own data config and modify the `max_action_dim` in:

https://github.com/NVIDIA/Isaac-GR00T/blob/17a77ebf646cf13460cdbc8f49f9ec7d0d63bcb1/gr00t/experiment/data_config.py#L127-L132

vitaliuwenting, 2025-07-25:
Has anyone tried revising the data config? it does not work for me. I got error below.

Traceback (most recent call last):
  File "/workspace/notebook/fine-tune-fih/gr00t_finetune.py", line 318, in <module>
    main(config)
  File "/workspace/notebook/fine-tune-fih/gr00t_finetune.py", line 290, in main
    experiment.train()
  File "/workspace/gr00t/experiment/runner.py", line 173, in train
    self.trainer.train(resume_from_checkpoint=self.resume_from_checkpoint)
  File "/workspace/gr00t/experiment/trainer.py", line 153, in train
    return super().train(resume_from_checkpoint, trial, ignore_keys_for_eval, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2245, in train
    return inner_training_loop(
           ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2560, in _inner_training_loop
    tr_loss_step = self.training_step(model, inputs, num_items_in_batch)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 3736, in training_step
    loss = self.compute_loss(model, inputs, num_items_in_batch=num_items_in_batch)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/experiment/trainer.py", line 75, in compute_loss
    outputs = model(inputs)
              ^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1736, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1747, in _call_impl
    return forward_call(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/accelerate/utils/operations.py", line 823, in forward
    return model_forward(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/accelerate/utils/operations.py", line 811, in __call__
    return convert_to_fp32(self.model_forward(*args, **kwargs))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/amp/autocast_mode.py", line 44, in decorate_autocast
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/model/gr00t_n1.py", line 168, in forward
    backbone_inputs, action_inputs = self.prepare_input(inputs)
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/model/gr00t_n1.py", line 186, in prepare_input
    self.validate_inputs(inputs)
  File "/workspace/gr00t/model/gr00t_n1.py", line 129, in validate_inputs
    raise ValueError(error_msg)
ValueError: Error: unexpected input/output
action.shape=torch.Size([2, 16, 49])

heyufan1995, 2025-07-28:
> Has anyone tried revising the data config? it does not work for me. I got error below.
> 
> Traceback (most recent call last): File "/workspace/notebook/fine-tune-fih/gr00t_finetune.py", line 318, in main(config) File "/workspace/notebook/fine-tune-fih/gr00t_finetune.py", line 290, in main experiment.train() File "/workspace/gr00t/experiment/runner.py", line 173, in train self.trainer.train(resume_from_checkpoint=self.resume_from_checkpoint) File "/workspace/gr00t/experiment/trainer.py", line 153, in train return super().train(resume_from_checkpoint, trial, ignore_keys_for_eval, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2245, in train return inner_training_loop( ^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2560, in _inner_training_loop tr_loss_step = self.training_step(model, inputs, num_items_in_batch) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 3736, in training_step loss = self.compute_loss(model, inputs, num_items_in_batch=num_items_in_batch) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/workspace/gr00t/experiment/trainer.py", line 75, in compute_loss outputs = model(inputs) ^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1736, in _wrapped_call_impl return self._call_impl(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/torch/nn/modules/module.py", line 1747, in _call_impl return forward_call(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/accelerate/utils/operations.py", line 823, in forward return model_forward(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/accelerate/utils/operations.py", line 811, in **call** return convert_to_fp32(self.model_forward(*args, **kwargs)) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/opt/conda/lib/python3.11/site-packages/torch/amp/autocast_mode.py", line 44, in decorate_autocast return func(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^ File "/workspace/gr00t/model/gr00t_n1.py", line 168, in forward backbone_inputs, action_inputs = self.prepare_input(inputs) ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/workspace/gr00t/model/gr00t_n1.py", line 186, in prepare_input self.validate_inputs(inputs) File "/workspace/gr00t/model/gr00t_n1.py", line 129, in validate_inputs raise ValueError(error_msg) ValueError: Error: unexpected input/output action.shape=torch.Size([2, 16, 49])