NVIDIA/Isaac-GR00T #82: Gr00t running on SO-100 arm, but it fails to do anything.

johnny555, 2025-04-01:
### 🐛 Describe the bug

Hi There, 

Thanks for the cool VLA model. I'm trying to get it to work on my SO-100 arm, and following the guide, but it is not working. 

I suspect I have missed some important key step, but I don't know what it is.

The robot arm initialises but then just shudders (see video)

https://github.com/user-attachments/assets/37b0f5a7-25e9-4051-ae61-894683d152d3

I followed the guide here: https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/3_new_embodiment_finetuning.ipynb to perform fine tuning, on this dataset: https://huggingface.co/datasets/badwolf256/so100_twin_cam_duck_v3 although only using the realsense scene cam as I didn't have VRAM for both video streams (using a 3090). 

I used this `modality.json` file: 
```
{
    "state": {
        "single_arm": {
            "start": 0,
            "end": 5
        },
        "gripper": {
            "start": 5,
            "end": 6
        }
    },
    "action": {
        "single_arm": {
            "start": 0,
            "end": 5
        },
        "gripper": {
            "start": 5,
            "end": 6
        }
    },
    "video": {
        "webcam": {
            "original_key": "observation.images.realsense"
        }
    },
    "annotation": {
        "human.task_description": {
            "original_key": "task_index"
        }
    }
}
```

It seemed to have converged after two runs based on my tensorboard output: 

[image]

I then setup an inference server using the command: 
`python scripts/inference_service.py --server     --model_path ~/so100-checkpoints/     --embodiment_tag new_embodiment     --data_config so100     --denoising_steps 4`

I then used this script to control the robot (using lerobot environment)  :

https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/examples/eval_gr00t_so100.py with some modifications (namely changing the text prompt, adjusting the RGB colours because they seemed wrong). 

I also needed the script https://github.com/NVIDIA/Isaac-GR00T/blob/main/gr00t/eval/service.py in the same directory that eval_groot_so100.py was run from.

There aren't any errors being output. The system appears to be running fine, but the model is not attempting the task. 

What have a missed ? Maybe I need more data? Do I need to train even longer? I'm training and inferencing on a 3090. 

Thanks for your help! 

### Versions
[... 474 lines cut ...]
torchvision==0.21.0
tornado==6.4.2
tqdm==4.67.1
traitlets==5.14.3
transformers==4.49.0
triton==3.2.0
txt2tags==3.9
types-python-dateutil==2.9.0.20241206
typing-inspect==0.9.0
typing_extensions==4.12.2
tzdata==2025.1
uri-template==1.3.0
urllib3==2.3.0
wandb==0.19.8
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
Werkzeug==3.1.3
widgetsnbextension==4.0.13
wrapt==1.17.2
xxhash==3.5.0
yarl==1.18.3
zarr==2.18.3
zipp==3.21.0

johnny555, 2025-04-04:
So it turns out that my model was just under fitted. 

After collecting a larger dataset (50 examples) and running on a large GPU (40GB VRAM) overnight I was able to get this working. I didn't use LoRA in the end.

Notably my grad norm was much smoother on the big GPU, I think training without LoRA is a good idea.

closed 2025-04-04 by johnny555

youliangtan, 2025-04-04:
thanks for updating your results. Normally we would use at least 40 - 50 trajectories for post training, or even larger (> 100), depending on the tasks. Since so100 is an unseen embodiment, it normally requires more trajectories.

VoicAndrei, 2025-04-17:
> So it turns out that my model was just under fitted.
>
> After collecting a larger dataset (50 examples) and running on a large GPU (40GB VRAM) overnight I was able to get this working. I didn't use [...]

Did you use both cameras during the overnight training? Also, could you share your training parameters for the 40GB VRAM GPU? Thanks!