NVIDIA/Isaac-GR00T #84: Time issues

Iam-Jane, 2025-04-01:
### 📚 The doc issue

I ran the inference code in practice, and it takes about 500ms to infer a single sample with a batch size of 1. Why is it so much slower?

youliangtan, 2025-04-02:
What compute are u using, and also image size. Pls provide more details

Iam-Jane, 2025-04-07:
Thank you for your response! I am using an NVIDIA 4090 GPU, and the image size is 224×224. The dataset  is ./demo_data/robot_sim.PickNPlace. However, I noticed that the line ”action_chunk = policy.get_action(dataset[0])“ takes approximately 500ms to execute.

youliangtan, 2025-04-07:
Thanks for reporting, i quickly tested on my side on a6000 pc 

```
Time taken to apply VLM transform: 0.12133550643920898 seconds
Time taken to apply all transforms: 0.25996851921081543 seconds
Time taken to get action from normalized input: 0.05869340896606445 seconds
Time taken to get unnormalized action: 0.0004832744598388672 seconds
Time taken to squeeze: 6.4373016357421875e-06 seconds
-> Total time to get action: 0.31922054290771484 seconds
```

Notice that the "transforms" process is taking most of the time. IIRC this seems to be a regression as it should be faster from our internal testing. But from the result above, the network forward pass is pretty quick, taking around 0.06 secs.

cc. @HuFY-dev

youliangtan, 2025-04-10:
I reran the benchmark with the latest code. Seems to be fine again. 🤔 

```
Time taken to unsqueeze: 3.457069396972656e-05 seconds
Time taken to apply transforms: 0.0034263134002685547 seconds
Time taken to get action from normalized input: 0.04645347595214844 seconds
Time taken to get unnormalized action: 0.008965730667114258 seconds
Time taken to squeeze: 7.3909759521484375e-06 seconds
--> Overall time taken to get action: 0.05893397331237793 seconds
```

The above inference is on a a6000 PC. Also, only use the second inference time number as the first pass normally takes longer due to jit

Iam-Jane, 2025-04-14:
I see, that's true. Thanks !

closed 2025-04-14 by Iam-Jane