NVIDIA/Isaac-GR00T #87: Question about how state token is handled for data sources without robot state (e.g. human video, neural trajectories)

jeonghopark07, 2025-04-03:
Hi, thank you for sharing this work on GR00T N1.

I have a question about how the model handles robot state information q_t during pre-training on data sources that do not provide such state inputs - like human videos or neural trajectories.

From the paper, I understand that the model expects q_t (robot proprioceptive state) as an input alongside image and language tokens. However, in the case of certain data (human video and neural trajectories), there is no associated robot state. I see that these examples are treated as a separate embodiment, but it's unclear from the paper.

What exactly is fed into the q_t input slot of the model when training on those data?
It would be great to get clarification on how q_t is handled in these cases.

Thanks in advance.

joeljang, 2025-04-07:
Hey Jeongho, thanks for the interest in GR00T N1. We just fill in 0's for the robot state information during pre-training for human videos & neural trajectories. We found empiricially that this does not deteriorate performance!

jeonghopark07, 2025-04-13:
Thanks @joeljang for your response. That answered my question.  
I'll go ahead and close the issue.

closed 2025-04-13 by jeonghopark07