NVIDIA/Isaac-GR00T #94: Question: Poor Results in Pick-Up Task Using NVIDIA Isaac GR00T N1 in Isaac Sim

KindenKuo, 2025-04-07:
I would like to quickly understand the capabilities of NVIDIA’s Isaac GR00T N1. Is this model ready to use directly with the GR1 robot, or does it require fine-tuning before deployment? I attempted to execute a simple pick-up task with it in Isaac Sim, but the performance was subpar.

youliangtan, 2025-04-07:
GR00t N1 is a pretrained model, and it should be used by finetuning. You technically can still use GR1 embodiment tag during posttraining which makes it faster to converge.

Jerryisqx, 2025-04-09:
> I would like to quickly understand the capabilities of NVIDIA’s Isaac GR00T N1. Is this model ready to use directly with the GR1 robot, or does it require fine-tuning before deployment? I attempted to execute a simple pick-up task with it in Isaac Sim, but the performance was subpar.

Hi, could you provide a quick note on how to evaluate GR1's task in Issac Sim environment?

youliangtan, 2025-04-10:
We will release our IsaacSim and Mujoco envs for eval in a few months, this would be easier for users to reproduce the results. Stay tuned

closed 2025-04-10 by youliangtan

wolfcanli, 2025-04-21:
> I would like to quickly understand the capabilities of NVIDIA’s Isaac GR00T N1. Is this model ready to use directly with the GR1 robot, or does it require fine-tuning before deployment? I attempted to execute a simple pick-up task with it in Isaac Sim, but the performance was subpar.

Hi, could you provide a quick note on how to evaluate GR1's task in Issac Sim environment? THANKS

wyt1234, 2025-05-21:
The same approach didn’t work. We’ve tried many tasks in Isaac Sim & Isaac Lab, but none were successful.