NVIDIA/Isaac-GR00T #101: Successfully finetune the model on RTX5090 and some tips

RX-02333, 2025-04-08:
I successfully finetune the model on RTX5090 and here are some tips  may help you:
Firstly,I use cuda 12.8 with ubuntu22.04,driver version570.133.07
Before install flash-attention,you should replace torch to nightly version with cuda12.8,then clone the flash-attention repo and build from source.It's important to change torch version before build.You cloud use MAX_JOBS to avoid out of memory issue.I set to 3 and it use 34G RAM peak.
After you successfully build it,you can finetune the model!
If you use RTX5090 ,here is a additionly tip.You should set finetune batch_size to 8 in scripts/gr00t_finetune.py to avoid out of memory.It use about 31G of VRAM,almost 32G VRAM of RTX5090.(kill other app witch use VRAM)
and here is my proof:

[image]

Chojins, 2025-04-09:
I wonder what the minimum vram requirement is for fine-tuning if MAX_JOBS and batch_size were decreased further and perhaps other parameters are tuned

RX-02333, 2025-04-09:
> I wonder what the minimum vram requirement is for fine-tuning if MAX_JOBS and batch_size were decreased further and perhaps other parameters are tuned

Hello,MAX_JOBS only cares about RAM when build the flash_attention library, and I try to set batch_size to 1 get the minimum
VRAM usage, but it also use 31G VRAM and gpu load  is lower.I guess it will allocate  all of available VRAM.So I can't measure the minimum VRAM usage, just kwon batch_size set to 8 can avoid OOM issue for 5090 .

kaushil24, 2025-04-09:
@RX-02333 this is great! To make it accessible to a larger audience, I'd suggest to summarize your findings to the [FAQ section](https://github.com/NVIDIA/Isaac-GR00T?tab=readme-ov-file#faq).

RX-02333, 2025-04-10:
> [@RX-02333](https://github.com/RX-02333) this is great! To make it accessible to a larger audience, I'd suggest to summarize your findings to the [FAQ section](https://github.com/NVIDIA/Isaac-GR00T?tab=readme-ov-file#faq).

Hello! I'm glad that this can help more people.Thank you and the excellent team!

closed 2025-08-28 by youliangtan