NVIDIA/Isaac-GR00T #98: Low Grasp Success Rate Despite Low Training Loss After LoRA Fine-Tuning on ALOHA Pick-and-Place Data

RuiboLiu77, 2025-04-08:
### 📚 The doc issue

I have fine-tuned the base model using the LoRA method on 100 pick-and-place trajectories from the ALOHA dataset, achieving a training loss below 0.005. When deploying the model in the real ALOHA environment, it exhibited strong error correction capabilities by quickly detecting whether an object was successfully picked up or placed, and by restarting the task as needed.

However, despite these positive aspects, the grasp success rate remains particularly low. The robotic arm appears to be limited in its grasping area compared to the broader range included in the training data. Consequently, when an object rotates or shifts slightly away, the model fails to grasp it correctly. 

I am curious whether this phenomenon could be attributed to the limited amount of training data or if other factors might be responsible. I would appreciate any insights or suggestions to improve the overall performance of the model.

Thank you~!

kaushil24, 2025-04-09:
Hi @RuiboLiu77 , there could be multiple factors affecting the grasp performance. For starters as you pointed out, the training data might not be diverse enough. Furthermore, does the camera position for inference resembles that during training; how different is the inference environment from training?

RuiboLiu77, 2025-04-11:
Thank you for your help! Here are the details about my workspace: I placed three cubes in different colors (red, orange, and blue) on a desk and I am trying to teach a robot how to pick up the closest one and place it into a box. To do this, I recorded 100 trajectories (33 trajectories per cube, plus one extra) from ALOHA.

Regarding your question about how different the inference environment is from training, I did not change any setup in my inference environment. The only modification I made was to slightly move and rotate the cubes (while still keeping them within the work range of the training data). However, the model could not successfully complete the task.

**P.S.** After I increased the dataset to 200 trajectories, the model performed better and could consistently complete the pick-and-place task on the first try. However, when I returned the cube to the workspace, the model was unable to locate the new cube or place it into the box.

youliangtan, 2025-04-18:
According to your description, it seems to be a data problem, in which your trajectory is doesnt not consists of the subsequent traj after the first task is completed. Usually for data collection you want to cover all states

I will close this ticket for now. If u have any more questions feel free to reopen this or comment below

closed 2025-04-18 by youliangtan