NVIDIA/Isaac-GR00T #99: Unable to Access single_panda_gripper Datasets via Hugging Face CLI and Python Script

pmh9960, 2025-04-08:
### 🐛 Describe the bug

Dear team,  
  
Thank you for your outstanding work and for sharing this impressive dataset.  
  
I have been attempting to download the `single_panda_gripper` datasets, but encountered consistent issues across both the Hugging Face CLI and Python API. For example, when executing the following CLI command:

```bash
huggingface-cli download nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim \
    --repo-type dataset \
    --include "single_panda_gripper.CloseDoubleDoor/**" \
    --local-dir ${HOME}/gr00t_dataset
```

the response indicates that no files were fetched:

```bash
Fetching 0 files: 0it [00:00, ?it/s]
```

I have also tested other subdirectories within `single_panda_gripper`, as well as a Python-based approach, but the issue persists specifically with this group of datasets. In contrast, other datasets under the same repository are downloadable without any problems.  
  
Given this consistent behavior, it appears that the issue may be specific to the `single_panda_gripper` subset. I would greatly appreciate it if you could investigate this matter when convenient.  
  
Thank you very much for your time and support.

### Versions

Python 3.10.16

huggingface-hub==0.29.3

youliangtan, 2025-04-09:
Hi, similiar issue here: https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/discussions/3

i also tried
```
huggingface-cli download nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim --repo-type dataset --include "gr1_full_upper_body**" --local-dir $HOME/gr00t_dataset
```
and it also doesnt work. Im suspecting this is a huggingface issue. im not too sure

cc. @loicmagne

dstx123, 2025-04-21:
> Hi, similiar issue here: https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/discussions/3
> 
> i also tried on a non panda dataset: `gr1_full_upper_body`
> 
> ```
> huggingface-cli download nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim --repo-type dataset \
>   --include "gr1_full_upper_body**"\
>  --local-dir $HOME/gr00t_dataset
> ```
> 
> and it also doesnt work. Im suspecting this is a huggingface issue. im not too sure
> 
> cc. [@loicmagne](https://github.com/loicmagne)

Hello. Is there any other way to obtain or create the single_panda_gripper dataset by myself? I look forward to your reply.

ShayekhBinIslam, 2025-04-26:
Try this with `git-lfs` installed:
```sh
git clone https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim
```

Or let me know if you require any subfolders, I have full 1.5 TB data downloaded.

dstx123, 2025-04-27:
> Try this with `git-lfs` installed:
> 
> git clone https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim
> Or let me know if you require any subfolders, I have full 1.5 TB data downloaded.

In this way I successfully downloaded single_panda_gripper. Thanks for your help.

closed 2025-06-15 by HuFY-dev

youliangtan, 2025-07-11:
Thanks @ShayekhBinIslam 

Another way is to download with github lfs: updated the comment here  https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/discussions/4

yanan1116, 2025-11-17:
```bash

hf download nvidia/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim \
--repo-type dataset \
--include "single_panda_gripper.CoffeePressButton/**" \
--local-dir /home/yanan/robotics/Isaac-GR00T/gr00t_dataset_official-CoffeePressButton

```

it hangs and no reponse for a long time.