NVIDIA/Isaac-GR00T #104: Reinforcement learning with GROOT N1

Vovinsa, 2025-04-09:
I have been reviewing the repository and the associated work on GR00T N1, and I would like to inquire about the integration of reinforcement learning (RL) within the project. Specifically, I am interested in whether the repository includes any ready-to-use RL components related to the training or fine-tuning of the control modules, or if the current implementation solely relies on imitation learning techniques without any built-in RL support.

If there are no such components currently available, could you please advise if the intended approach requires a completely manual setup of RL mechanisms? Additionally, if a manual configuration is anticipated, I would greatly appreciate it if you could point me toward any existing resources, external repositories, or documentation that provide a suitable foundation or reference for integrating RL within this context.

kaushil24, 2025-04-09:
Hi @Vovinsa , the gr00t repo currently only uses imitation learning via the N1 model, there is no RL integration yet. 

>If there are no such components currently available, could you please advise if the intended approach requires a completely manual setup of RL mechanisms?

Could you explain how do you intend to use RL policy in conjuncture with N1?

Vovinsa, 2025-04-09:
> Hi [@Vovinsa](https://github.com/Vovinsa) , the gr00t repo currently only uses imitation learning via the N1 model, there is no RL integration yet.
>
>
> Could you explain how do you intend to use RL [...]

Thank you for your follow-up. I am exploring the integration of RL policy alongside the N1 model in order to address scenarios where pre-collected training data may be limited or insufficient for certain tasks. By incorporating RL, I aim to enable the robot to adapt more quickly to novel tasks and to improve its performance in tasks where the current behavior still shows inconsistencies.
In practical terms, the idea is to allow the robot to interact directly with its environment, collect feedback or rewards, and iteratively update its policy without relying solely on large supervised datasets

kaushil24, 2025-04-10:
That's an interesting thread imo. Currently, we don't have any RL mechanism in place. Do let me know if you need any help in navigating how to integrate.

BouajilaHamza, 2025-05-10:
@Vovinsa i have been thinking of the same thing .
It would be great to share ideas .
The first intuition i got is to keep the same logic of the policy .
Export each model llm , eagle backbone and the action had separately into onnx format and combine them using a policy logic same as this one .

SakethTecholution, 2025-11-06:
May be we should look on how these reasoning llms work , because there both supervised leaning models and rl mechanism is intact