NVIDIA/Isaac-GR00T #111: ValueError: Error occurred in initializing dataset robot_sim.PickNPlace:

Luo-Zhongwei, 2025-04-10:
### 🐛 Describe the bug

hi,I follow your readme to reproduce the demo and get such error after finetune 
A:  I follow this  to fine tune
```python
python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 1
```

B : and I think get quite great log 
```python
{'loss': 0.0085, 'grad_norm': 0.19532835483551025, 'learning_rate': 6.834750376549792e-09, 'epoch': 150.76}    
{'loss': 0.0025, 'grad_norm': 0.12665756046772003, 'learning_rate': 4.3742761183018784e-09, 'epoch': 150.91}   
{'loss': 0.0022, 'grad_norm': 0.08897408843040466, 'learning_rate': 2.4605460129556445e-09, 'epoch': 151.06}   
{'loss': 0.003, 'grad_norm': 0.39885759353637695, 'learning_rate': 1.0935809887702154e-09, 'epoch': 151.21}    
{'loss': 0.0024, 'grad_norm': 0.10645890235900879, 'learning_rate': 2.7339599464326627e-10, 'epoch': 151.36}   
{'loss': 0.0039, 'grad_norm': 0.11007785052061081, 'learning_rate': 0.0, 'epoch': 151.52}                      
{'train_runtime': 10338.0838, 'train_samples_per_second': 30.954, 'train_steps_per_second': 0.967, 'train_loss': 0.014880996694974601, 'epoch': 151.52}
100%|██████████████████████████████████████████████████████████████████| 10000/10000 [2:52:15<00:00,  1.03s/it]
wandb: 🚀 View run /root/autodl-tmp/Isaac-GR00T/all_ckpt/toy/ at: https://wandb.ai/485434977-shang-hai-jiao-tong/huggingface/runs/vjj6euxm
wandb: Find logs at: wandb/run-20250411_020736-vjj6euxm/logs
```

C: then i use this command to eval 👍 
```python
~/autodl-tmp/Isaac-GR00T# python scripts/eval_policy.py --plot --model_path /root/autodl-tmp/Isaac-GR00T/all_ckpt/toy --embodiment_tag new_embodiment
```

then get such errror

```python
Loading pretrained dual brain from /root/autodl-tmp/Isaac-GR00T/all_ckpt/toy
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /root/autodl-tmp/Isaac-GR00T/all_ckpt/toy
Total number of DiT parameters:  537803776
Tune action head projector: True
Tune action head diffusion model: True
Loading checkpoint shards: 100%|█████████████████████████████████████████████████| 2/2 [00:01<00:00,  1.12it/s]
Tune action head projector: True
Tune action head diffusion model: True
{'video': ModalityConfig(delta_indices=[0], modality_keys=['video.ego_view']), 'state': ModalityConfig(delta_indices=[0], modality_keys=['state.left_arm', 'state.right_arm', 'state.left_hand', 'state.right_hand', 'state.waist']), 'action': ModalityConfig(delta_indices=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], modality_keys=['action.left_arm', 'action.right_arm', 'action.left_hand', 'action.right_hand', 'action.waist']), 'language': ModalityConfig(delta_indices=[0], modality_keys=['annotation.human.coarse_action'])}
Initialized dataset robot_sim.PickNPlace with new_embodiment
Traceback (most recent call last):
  File "/root/autodl-tmp/Isaac-GR00T/gr00t/data/dataset.py", line 448, in _check_integrity
    self.lerobot_modality_meta.get_key_meta(key)
  File "/root/autodl-tmp/Isaac-GR00T/gr00t/data/schema.py", line 162, in get_key_meta
    raise ValueError(
ValueError: Key: annotation.human.coarse_action, annotation key human.coarse_action not found in metadata, available annotation keys: dict_keys(['human.action.task_description', 'human.validity'])

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/autodl-tmp/Isaac-GR00T/scripts/eval_policy.py", line 101, in <module>
    dataset = LeRobotSingleDataset(
  File "/root/autodl-tmp/Isaac-GR00T/gr00t/data/dataset.py", line 151, in __init__
    self._check_integrity()
  File "/root/autodl-tmp/Isaac-GR00T/gr00t/data/dataset.py", line 450, in _check_integrity
    raise ValueError(
ValueError: Error occurred in initializing dataset robot_sim.PickNPlace:
Unable to find key annotation.human.coarse_action in modality metadata:
Key: annotation.human.coarse_action, annotation key human.coarse_action not found in metadata, available annotation keys: dict_keys(['human.action.task_description', 'human.validity'])

```

### Versions

```python

[... 141 lines cut ...]
tensorflow-io-gcs-filesystem 0.37.1
termcolor                    3.0.1
tianshou                     0.5.1
tifffile                     2025.3.30
timm                         1.0.14
tokenizers                   0.20.3
tomli                        2.2.1
torch                        2.5.1
torchvision                  0.20.1
tqdm                         4.67.1
transformers                 4.45.2
triton                       3.1.0
typeguard                    4.4.2
typing_extensions            4.12.2
tyro                         0.9.17
tzdata                       2025.2
urllib3                      2.3.0
wandb                        0.18.0
Werkzeug                     3.1.3
wheel                        0.45.1
wrapt                        1.14.1
yacs                         0.1.8
zipp                         3.21.0
zmq                          0.0.0
```

caijianwei1996, 2025-04-11:
seems like this demo data not have gt

youliangtan, 2025-04-18:
Im not able to reproduce this issue:

with the scripts below

```bash
python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 1  --lora-rank 16   --save-
python scripts/eval_policy.py --plot --model_path /tmp/gr00t/checkpoint-10/ --embodiment_tag new_embodiment
```

I will close this for now, reopen it if you encounter it again

closed 2025-04-18 by youliangtan

Luo-Zhongwei, 2025-04-20:
> Im not able to reproduce this issue:
> 
> with the scripts below
> 
> python scripts/gr00t_finetune.py --dataset-path ./demo_data/robot_sim.PickNPlace --num-gpus 1  --lora-rank 16   --save-
> python scripts/eval_policy.py --plot --model_path /tmp/gr00t/checkpoint-10/ --embodiment_tag new_embodiment
> I will close this for now, reopen it if you encounter it again

WHY NOT ... it from your readme