NVIDIA/Isaac-GR00T #114: Too much randomness in robot movements

1g0rrr, 2025-04-11:
### 🐛 Describe the bug

Hi! Thanks for the super impressive work!

I see a lot of randomness in robot movements and it looks shaky because of this. With pi0 I dont have this issue.  Is this supposed behavior? Is there a way to smooth it out?
Attached the video with 40k steps of training.

https://github.com/user-attachments/assets/50b910f8-56ad-48a8-abf7-1e6e33ce0728

### Versions

version from installation instructions

Catchip, 2025-04-12:
From what I've read on a Chinese tech blog, it mentioned dealing with robotic shaking arm issue. The blogger apparently resolved the shaking problem by reducing the PID coefficient. Hope this helps，just sharing as a  beginner.

1g0rrr, 2025-04-12:
> From what I've read on a Chinese tech blog, it mentioned dealing with robotic shaking arm issue. The blogger apparently resolved the shaking problem by reducing the PID coefficient. Hope this helps，just sharing as a beginner.

Added Pi0 run for comparison. So robot itself don’t shake

youliangtan, 2025-04-13:
Thanks benchmark our Gr00t N1 against other models. On top of my head, could you try to increase the [denoising_steps](https://github.com/NVIDIA/Isaac-GR00T/blob/40070999be99dcbc255dba6fa323cbcb2fafa9b9/scripts/inference_service.py#L54) during inference? (increase to 10 or even up to 16) See if this helps

Catchip, 2025-08-26:
@TilTheunissen I recently found that the jitter situation improves significantly with the time ensemble method.
A sample implementation
```python
class TimeEnsembler:
    def __init__(self, balancing_factor=0.1, horizon=30, n_joints=16):
        self.balancing_factor = balancing_factor
        self.n_joints = n_joints
        self.temporal_size = int(horizon)
        self.temporal_mask = torch.flip(
            torch.triu(
                torch.ones(self.temporal_size, self.temporal_size, dtype=torch.bool)
            ),
            dims=[1],
        ).numpy()

        self.action_buffer = np.zeros(
            (self.temporal_mask.shape[0], self.temporal_mask.shape[0], self.n_joints)
        )
        self.action_buffer_mask = np.zeros(
            (self.temporal_mask.shape[0], self.temporal_mask.shape[0]), dtype=np.bool_
        )
        # Action chunking with temporal aggregation
        self.temporal_weights = np.array(
            [np.exp(-1 * balancing_factor * i) for i in range(self.temporal_size)]
        )[:, None]

    def ensemble(self, action_chunk):
        # action_chunk:[horizon, n_joints]

        # shift buffer
        self.action_buffer[1:, :, :] = self.action_buffer[:-1, :, :]
        self.action_buffer_mask[1:, :] = self.action_buffer_mask[
            :-1, :
        ]
        self.action_buffer[:, :-1, :] = self.action_buffer[:, 1:, :]
        self.action_buffer_mask[:, :-1] = self.action_buffer_mask[
            :, 1:
        ]
        self.action_buffer_mask = (
            self.action_buffer_mask * self.temporal_mask
        )

        # Add action chunk to action buffer
        self.action_buffer[0] = action_chunk
        self.action_buffer_mask[0] = np.array(
            [True] * self.temporal_mask.shape[0], dtype=np.bool_
        )

        # Ensemble temporally to predict next action
        action = np.sum(
            self.action_buffer[:, 0, :]
            * self.action_buffer_mask[:, 0:1]
            * self.temporal_weights,
            axis=0,
        ) / np.sum(
            self.action_buffer_mask[:, 0:1] * self.temporal_weights
        )

        return action  # (n_joints,)

```