NVIDIA/Isaac-GR00T #120: RoboCasa multi-task post-training - Performance reproduction

youthHan, 2025-04-13:
### 🐛 Describe the bug

I use the single-panda-gripper data available on HF for post-training. However, it is hard for me to reproduce the multi-task RoboCasa performance. I tried single-task training to validate that my evaluation steps and simulation setup is fine.

Roughly, I sample 100 demos per task and mixed the data per action step.

Stats: Min/Max of the newly mixed data is obtained directly retrieved and merged from each task meta file without recalculation.
Hyperparameter: batch size per device is set to 32, the learning rate is set to 1e-4, total 100k gradient steps, on an 8-A100 node
Data config: SinglePandaArmGripper config, which is available in the official codes and only reads [part of the available robot states](https://github.com/NVIDIA/Isaac-GR00T/blob/1184da1013ceb359f8b0e1801280ae12347396df/gr00t/experiment/data_config.py#L542). 
Counterpart models: I trained single-task models on CloseDrawer and PnPCounterToCab with sampled action steps.
RoboCasa: I am using the latest version.

I found out that in my experiment: Longer training, worse performance (I tested on CloseDrawer task, which is simple and compared ckpt-1500 and ckpt-84000). Though both ckpt resulted in 0 success rate, the ckpt-1500 at least produced reasonable action trajectory towards the drawer, while ckpt-84000 generated slow actions with unreasonable steps.

The performance on single-task CloseDrawer task: I tried 3 episodes with 2 succeeded.

My question: 
- What are the hyperparameters used for robocasa post-training? The experiment setup table in the paper only mentions an extensive range.
- What would be the cause of the bad reproduced performance? 

### Versions

Python 3.10.16
absl-py==2.2.1
accelerate==1.2.1
aiosignal==1.3.2
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
asttokens==3.0.0
astunparse==1.6.3
attrs==25.3.0
av==12.3.0
blessings==1.7
cachetools==5.5.2
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
click==8.1.8
cloudpickle==3.1.1
comm==0.2.2
contourpy==1.3.1
cramjam==2.9.1
cycler==0.12.1
Cython==3.0.12
debugpy==1.8.13
decorator==5.2.1
decord==0.6.0
diffusers==0.30.2
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.16
einops==0.8.1
eval_type_backport==0.2.2
exceptiongroup==1.2.2
executing==2.2.0
Farama-Notifications==0.0.4
fasteners==0.19
fastparquet==2024.11.0
filelock==3.18.0
flash-attn==2.7.1.post4
flatbuffers==25.2.10
fonttools==4.56.0
frozenlist==1.5.0
fsspec==2025.3.0
fvcore==0.1.5.post20221221
gast==0.6.0
gitdb==4.0.12
GitPython==3.1.44
glfw==2.8.0
google-auth==2.38.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
gr00t==0.1.0
grpcio==1.71.0
gym==0.26.2
gym-notices==0.0.8
gymnasium==1.0.0
h5py==3.12.1
huggingface-hub==0.29.3
hydra-core==1.3.2
idna==3.10
imageio==2.34.2
importlib_metadata==8.6.1
iniconfig==2.1.0
iopath==0.1.10
ipykernel==6.29.5
ipython==8.34.0
jedi==0.19.2
Jinja2==3.1.6
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter_client==8.6.3
jupyter_core==5.7.2
keras==2.15.0
kiwisolver==1.4.8
kornia==0.7.4
kornia_rs==0.1.8
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
mdurl==0.1.2
-e git+https://github.com/Robert-gyj/Prediction_with_Action.git@a0bdc12e05d435a77efd830ee6953d556c1a6083#egg=metaworld&subdirectory=metaworld
ml-dtypes==0.2.0
mpmath==1.3.0
msgpack==1.1.0
mujoco-py==2.1.2.14
nest-asyncio==1.6.0
networkx==3.4.2
numba==0.61.0
numpy==1.26.4
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.2.2
omegaconf==2.3.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
opt_einsum==3.4.0
packaging==24.2
pandas==2.2.3
parso==0.8.4
peft==0.14.0
pettingzoo==1.24.3
pexpect==4.9.0
pillow==11.1.0
pipablepytorch3d==0.7.6
platformdirs==4.3.7
pluggy==1.5.0
portalocker==3.1.1
prompt_toolkit==3.0.50
protobuf==3.20.3
psutil==7.0.0
ptyprocess==0.7.0
pure_eval==0.2.3
pyarrow==14.0.1
pyasn1==0.6.1
pyasn1_modules==0.4.2
pyav==14.2.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
Pygments==2.19.1
pyparsing==3.2.3
pytest==8.3.5
python-dateutil==2.9.0.post0
pytz==2025.2
PyYAML==6.0.2
pyzmq==26.3.0
ray==2.40.0
referencing==0.36.2
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
rich==13.9.4
rpds-py==0.24.0
rsa==4.9
safetensors==0.5.3
scikit-image==0.25.2
scipy==1.15.2
sentry-sdk==2.24.1
setproctitle==1.3.5
shtab==1.7.1
six==1.17.0
smmap==5.0.2
stack-data==0.6.3
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.5.0
tianshou==0.5.1
tifffile==2025.3.30
timm==1.0.14
tokenizers==0.20.3
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tornado==6.4.2
tqdm==4.67.1
traitlets==5.14.3
transformers==4.45.2
triton==3.1.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
urllib3==2.3.0
wandb==0.18.0
wcwidth==0.2.13
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp==3.21.0
zmq==0.0.0

Jerryisqx, 2025-04-14:
Hi, I'm also curious about the reproduce result. Could you share how to load the ckpt successfully on robocasa data? The post-training data seems unable to download. Also, the robocasa simulation needs .pth weights and original finetuning method uses .safetensor. Is there a file tranform needed?

Vovinsa, 2025-04-15:
Hi! Can you explain, how you have downloaded this dataset? And maybe do you have Urdf files of the single_panda_gripper model? Thank you

Vovinsa, 2025-04-17:
> @Jerryisqx @Vovinsa I directly download the full post-training data from HF and use single-panda-gripper data only. For the checkpoint in GR00T, I load the available one from HF.

Tank you! Have you made simulations with single panda gripper?

Jerryisqx, 2025-04-17:
@youthHan Thanks, can the finetuned gr00t checkpoint be directly loaded in robocasa envrionment? I saw the policy evaluation scripts here (https://robocasa.ai/docs/use_cases/policy_learning.html) I suppoer after finetuning in gr00t, we can directly use the checkpoint in this script?

youthHan, 2025-04-21:
@Jerryisqx @Vovinsa GR00T officially provides server-client mode for deployment, where the trained checkpoint should be loaded at the server side. You can write the client code of your own choice of simulator to capture observations, send to the server for prediction and apply the server-generated actions back to the simulator.

youthHan, 2025-05-07:
@Jinseong1126 Hi, not been much progress regarding this issue. I ended up with training 500k gradient steps; however, I failed to reproduce the performance reported in the paper (with a huge gap).  It could be a simulator version mismatch or a training recipe difference. However, I don't have further clues until we have official responses or guidelines.

Jerryisqx, 2025-06-29:
Hi @youthHan, I also tried for 64 batches per gpu on 2 a6000 finetuned for 15k steps and there are 50% fake fail (finished but report as failed). Did you also meet this issue?

xwjiang2010, 2025-09-16:
Please see https://github.com/NVIDIA/Isaac-GR00T/pull/357 for finetuning recipe on RoboCasa and also how to evaluate it in the simulation environment.