NVIDIA/Isaac-GR00T #124: GR1 dof order

Wonder1905, 2025-04-14:
Hi, you have mentioned in :
[modality.json](https://github.com/NVIDIA/Isaac-GR00T/blob/main/demo_data/robot_sim.PickNPlace/meta/modality.json)
A coarse dof order of gr1:
` "action": {
        "left_arm": {
            "start": 0,
            "end": 7
        },
        "left_hand": {
            "start": 7,
            "end": 13
        },
        "left_leg": {
            "start": 13,
            "end": 19
        },
        "neck": {
            "start": 19,
            "end": 22
        },
        "right_arm": {
            "start": 22,
            "end": 29
        },
        "right_hand": {
            "start": 29,
            "end": 35
        },
        "right_leg": {
            "start": 35,
            "end": 41
        },
        "waist": {
            "start": 41,
            "end": 44
        }
    },`

Is there any chance to share what is the internal order? i,e the pitch roll yaw?

Thanks a lot!

youliangtan, 2025-04-18:
@squarefk do you know the exact order here?

squarefk, 2025-04-18:
For each limb, it should follow the order [as shown here]. (https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/robots/gr1/robot.xml).

closed 2025-04-29 by youliangtan