NVIDIA/Isaac-GR00T #125: Regarding the latent actions mentioned in your paper

Yuheng2000, 2025-04-14:
May I ask if the latent actions generated by VQ-VAE during the pre-training phase have the same dimensionality as the final output from System 1? Additionally, for the "LAPA" embodiment, what constitutes the input robot state? Furthermore, can the latent actions learned by VQ-VAE and the IDM-labeled actions be used simultaneously to supervise the model's outputs?

Thank you very much for your clarification!

joeljang, 2025-04-24:
Yes, the latents have the same dimensionality as the final output. And yes, we are using the 'simultaneously' in some sense.

When training the VQ-VAE, we ensured that we are extracting a 512 dimension vector, which corresponds to 16 x 32, which is the action dimension used during the training (16 action chunks of 32). 

For one thing, latent actions are applied to ALL of the data (human, robot, simulated, neural). Secondly, for 'neural trajectories', we use IDM-labeled actions. So you can think of 'neural trajectories' as having two clones, one labeled with latent actions and one labeled with IDM actions. They are both used in the same way, being treated as a 'unique' embodiment.

Hope this clears things up!

WangXinYi111, 2025-07-08:
Thanks for sharing your excellent work. I would like to ask you about the latent actions in N1 paper, which mentioned that you  retrieve similar latent embeddings across various embodiments. How did you retrieve them? What indicators did you use to measure whether two instances are similar? Looking forward to your reply