NVIDIA/Isaac-GR00T #129: Question about evaluation of the simulation experiments

prilgrim-whtie, 2025-04-17:
### 📚 The doc issue

For the evaluation of the simulation experiments in the paper, can you provide some examples (simulation platform, environment settings, etc.), especially for the humanoid robot GR-1, rather than a simple MSE?

[image]

dstx123, 2025-04-21:
Looking forward to the relevant code being released.

RN-Tesseract, 2025-04-22:
Looking forward to the release of environments and code to test and train the model in simulation

youliangtan, 2025-09-09:
Libero: https://github.com/NVIDIA/Isaac-GR00T/pull/357
SimpleEnv: https://github.com/NVIDIA/Isaac-GR00T/tree/main/examples/SimplerEnv
Robocasa: https://github.com/robocasa/robocasa-gr1-tabletop-tasks