NVIDIA/Isaac-GR00T #130: Poor fine-tuning results on LeRobot SO100 with GR00T N1

VoicAndrei, 2025-04-19:
## Poor fine-tuning results on LeRobot SO100 with GR00T N1

### Issue Description:

I have built the LeRobot SO100 robotic arm specifically to test the GR00T N1 model. I've collected two distinct datasets aimed at a task involving picking up a banana and placing it on a plate:

- **[Dataset 1](https://huggingface.co/datasets/VoicAndrei/so100_banana_to_plate_rebel_full)**: 50 episodes with teleoperated actions to pick and place a banana, including various perturbations (e.g., apple, fork, changing banana positions).
- **[Dataset 2](https://huggingface.co/datasets/VoicAndrei/so100_banana_to_plate_only)**: 120 episodes, teleoperated similarly but with minimal perturbations—only banana and plate consistently present.

### Experiment Details & Issues:

Initially, extensive tuning attempts on my local RTX 4090 with LoRA yielded poor results, leading me to switch to an A100 GPU on the cloud. Despite this, fine-tuning outcomes remained disappointing:

- **Experiment 1 (10k steps, Dataset 1, top camera only)**: Surprisingly, yielded the best performance—robot successfully completed the task exactly once without assistance, but consistently failed on other attempts.
- **Experiment 2 (20k steps, Datasets 1+2 combined, both cameras)**: Extremely poor performance; robot exhibited jittering behavior without any meaningful task completion.
- **Experiment 3 (3k, 4.5k, 6k, 7.5k steps, Datasets 1+2 combined, top camera only)**: Generally poor results, slightly better than dual-camera. The robot approached the banana and closed the gripper but failed to pick and place the object.

### Questions & Support Needed:

1. Could this issue be related to the quality or structure of my datasets?
2. Are there specific hyperparameters or training strategies recommended for tasks similar to mine?

I've used default fine-tuning parameters. Below are the exact commands for:

- **Fine-tuning**:
  ```bash
  python scripts/gr00t_finetune.py \
   --dataset-path /datasets/so100_banana_to_plate_rebel_full/ \
   --num-gpus 1 \
   --output-dir ~/so100-banana  \
   --max-steps 20000 \
   --data-config so100 \
   --video-backend torchvision_av
  ```
- **Server inference node**:
  ```bash
  python scripts/inference_service.py \
  --server \
  --model_path ~/so100-banana/ \
  --embodiment_tag new_embodiment \
  --data_config so100 \
  --denoising_steps 4  # I also tried with 6 and higher numbers, same results
  ```
- **Client node**:
  ```bash
  python examples/eval_gr00t_so100.py \
  --use_policy \
  --host localhost \
  --port 5555 \
  --camera_index 0
  ```

### Additional Information:

Here are videos demonstrating the robot's behaviors:

- Here's the only time the robot completed the task (On Experiment 1)

https://github.com/user-attachments/assets/87bcf5c0-e812-4db0-974b-2e564cec1381

- Here's how the model usually performs on my robot:

https://github.com/user-attachments/assets/1aaa3eb9-66ca-4c4f-a54a-5c054382367a

https://github.com/user-attachments/assets/187203a4-4b8b-4a87-a71b-43d7fec7bf45

Any insights or guidance would be highly appreciated. Thanks in advance!

ziqiyang107, 2025-04-21:
We got the similar behavior, the gripper kept jittering around the object, without grasping, or sometimes, it did grasp, but then could not lift the object up and move it to another location

wuxiaoqiang12, 2025-04-24:
Hi, I'm tried to deploy the GR00T to lerobot so100 arm as you did, but I encounter some package confict, lerobot requirest  "gymnasium==0.29.1",   but GR00T requirest "gymnasium==1.0.0", and some other package conflict.
```shell
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
gr00t 0.1.0 requires gymnasium==1.0.0, but you have gymnasium 0.29.1 which is incompatible.
gr00t 0.1.0 requires opencv_python==4.8.0.74, but you have opencv-python 4.11.0.86 which is incompatible.
gr00t 0.1.0 requires pyarrow==14.0.1, but you have pyarrow 19.0.1 which is incompatible.
gr00t 0.1.0 requires torch==2.5.1, but you have torch 2.7.0 which is incompatible.
gr00t 0.1.0 requires torchvision==0.20.1, but you have torchvision 0.22.0 which is incompatible.
pettingzoo 1.25.0 requires gymnasium>=1.0.0, but you have gymnasium 0.29.1 which is incompatible.
```
Do you know how to solve this problem? I use conda to created a gr00t env, install gr00t first and then install lerobot into this env. Thank you very much.

flazerain, 2025-04-24:
> We got the similar behavior, the gripper kept jittering around the object, without grasping, or sometimes, it did grasp, but then could not lift the object up and move it to another location

We got the similar behavior.The best-case is the gripper keeps grabbing at air, with only occasional successful captures of the target.

wuxiaoqiang12, 2025-04-24:
> Hi, I'm tried to deploy the GR00T to lerobot so100 arm as you did, but I encounter some package confict, lerobot requirest "gymnasium==0.29.1", but GR00T requirest "gymnasium==1.0.0", and some other package conflict.你好，我尝试像你一样将 GR00T 部署到 lerobot so100 arm，但是我遇到了一些包冲突，lerobot 需要“gymnasium==0.29.1”，但是 GR00T 需要“gymnasium==1.0.0”，还有一些其他包冲突。
> 
> ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
> gr00t 0.1.0 requires gymnasium==1.0.0, but you have gymnasium 0.29.1 which is incompatible.
> gr00t 0.1.0 requires opencv_python==4.8.0.74, but you have opencv-python 4.11.0.86 which is incompatible.
> gr00t 0.1.0 requires pyarrow==14.0.1, but you have pyarrow 19.0.1 which is incompatible.
> gr00t 0.1.0 requires torch==2.5.1, but you have torch 2.7.0 which is incompatible.
> gr00t 0.1.0 requires torchvision==0.20.1, but you have torchvision 0.22.0 which is incompatible.
> pettingzoo 1.25.0 requires gymnasium>=1.0.0, but you have gymnasium 0.29.1 which is incompatible.
> Do you know how to solve this problem? I use conda to created a gr00t env, install gr00t first and then install lerobot into this env. Thank you very much.你知道怎么解决这个问题吗？我用 conda 创建了一个 gr00t 环境，先安装 gr00t，然后再把 lerobot 安装到这个环境中。非常感谢。

I solved this by modify the gr00t requires of package version

wuxiaoqiang12, 2025-04-24:
> Hi, I'm tried to deploy the GR00T to lerobot so100 arm as you did, but I encounter some package confict, lerobot requirest "gymnasium==0.29.1", but GR00T requirest "gymnasium==1.0.0", and some other package conflict.
> 
> ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
> gr00t 0.1.0 requires gymnasium==1.0.0, but you have gymnasium 0.29.1 which is incompatible.
> gr00t 0.1.0 requires opencv_python==4.8.0.74, but you have opencv-python 4.11.0.86 which is incompatible.
> gr00t 0.1.0 requires pyarrow==14.0.1, but you have pyarrow 19.0.1 which is incompatible.
> gr00t 0.1.0 requires torch==2.5.1, but you have torch 2.7.0 which is incompatible.
> gr00t 0.1.0 requires torchvision==0.20.1, but you have torchvision 0.22.0 which is incompatible.
> pettingzoo 1.25.0 requires gymnasium>=1.0.0, but you have gymnasium 0.29.1 which is incompatible.
> Do you know how to solve this problem? I use conda to created a gr00t env, install gr00t first and then install lerobot into this env. Thank you very much.

I solved this issue by modifing the gr00t requires of packages.

huliang999, 2025-04-25:
same poor fine-tuning performance in aloha sim and robotwin sim

DawnyWu, 2025-05-03:
same poor performance with 2 cameras

youliangtan, 2025-05-12:
@VoicAndrei Thanks for the detailed experimentation! the robot seems to struggle at "looking" at the banana.

Fyi, I do observe lower performance with LoRa. 

Also how are you using multiple dataset? Which normalization stats are you using when training, because norm stats which determines how state and action are normalized is extremely important during multi dataset training. 

Also for camera ablation. I would think adding additional camera can yield a lower loss when training, did you observe that when doing a fair comparison?

VoicAndrei, 2025-05-12:
> [@VoicAndrei](https://github.com/VoicAndrei) Thanks for the detailed experimentation! the robot seems to struggle at "looking" at the banana.
>
> Fyi, I do observe lower performance with LoRa.
>
> Also [...]

---
@youliangtan Hello!

I've since shifted away from the banana experiment to focus on 3D printing small cubes. In this new setup, I employed three cameras, with the wrist camera proving most impactful. You can view the dataset [here](https://huggingface.co/datasets/VoicAndrei/so100_cubes_three_cameras) (150 episodes).

Due to VRAM limitations—even on an A100 40GB GPU—I fine-tuned the model using only the top and wrist cameras. Despite using the exact same fine-tuning, server, and client scripts as before, this setup resulted in significantly improved success rates. It even discriminates the cubes based on color (red vs green) when instructed to do so (but only sometimes). Do you think that for better discrimination I should try with different objects instead of colors? (e.g., banana vs apple).

Also, if I want to train my SO100 to also be able to open a drawer on another 150 episodes, should I fine-tune a single model with all 300 episodes (150 from cubes + 150 drawer episodes)?

youliangtan, 2025-05-12:
@VoicAndrei Nice that you got that working.

In terms of language following, what's the success rate right now? My immediate feedback after looking at the dataset is to have more mixed "red" and "green" objects on the plate (which u are doing, consists of approx 20% of the dataset). Or you can try to label the "only" dataset with a "all green/red cube" and try if that works better.

in terms of multitask policy training + a drawer task. it is possible to train your policy with 2 distinct dataset, but i think it wouldnt see much improvements since the tasks are quite different.

wolfcanli, 2025-06-04:
@huliang999 @VoicAndrei @flazerain @ziqiyang107 Could you help us identify the root cause of this issue? We trained our model using 150 physical robot demonstration trajectories for cup-stacking tasks. While the offline dataset evaluation results show some variance (performance curve not perfectly smooth), the predicted joint command trends are generally correct. However, during physical deployment testing, the robotic arm exhibits persistent oscillations at its initial position. Thanks.

[image]

https://github.com/user-attachments/assets/c8e1448a-56e6-4ccc-80dd-64a078d7420d

aol-work, 2025-06-10:
fwiw I struggled with similar problems as in the original Issue Description and for me although a simplistic SO-101 fine tune (20 examples, 10s episodes, top camera only, "shift the red cube towards the grey mark") worked in the end all the fine tunes I have tried seem to be very sensitive to small changes in top camera position or camera angle. It worked well when there was no deviation at all between examples and deployment, but it stopped working entirely when I shifted the camera only by 1cm. As expected, this seems to improve with each round of added examples with different camera positions, although I have not reached a point where the model generalizes to cover small perturbations. After 3 more rounds of examples the model for the simple task above still struggles with finding correct end effector positions. Anyways, if a simple toy example does not work for you, double check the camera did not move.

yuanlin2004, 2025-08-05:
> [@VoicAndrei](https://github.com/VoicAndrei) Thanks for the detailed experimentation! the robot seems to struggle at "looking" at the banana.
>
> Fyi, I do observe lower performance with LoRa.
>
> Also [...]

Hi @youliangtan , I've also encountered bad results with my fine-tuned results. Could you please elaborate on the normalization part? Is it some extra step or tools argument we can set? I was following https://huggingface.co/blog/nvidia/gr00t-n1-5-so101-tuning, which does not seem to mention the normalization step. 

Thanks!

Yuan