NVIDIA/Isaac-GR00T #139: state mapping between dataset and simulator (dexmimicgen)

yfpeng1234, 2025-04-27:
Dear author,
Thanks for open-sourced this interesting work. I try to test the success rate of grootn1 in dexmimicgen benchmark. However, I found the definition of state in your dataset is:

[image]
But in dexmimicgen, the definition of state is:

[image]
Could you please share how can I map the state in simulator the dataset format so that it can be fed to grootn1. By the way, should I modify the action or image observation.
Best

xwjiang2010, 2025-09-17:
@yfpeng1234 
Please follow this [MR](https://github.com/NVIDIA/Isaac-GR00T/pull/357/files) to see how to plug in RoboCasa sim env.
I will close this for now but feel free to ping back if any issues.

Huaizz-shawen, 2025-09-18:
> [@yfpeng1234](https://github.com/yfpeng1234) Please follow this [MR](https://github.com/NVIDIA/Isaac-GR00T/pull/357/files) to see how to plug in RoboCasa sim env. I will close this for now but feel free to ping back if any issues.

Thanks for the answer for this issue! But i found no more change about robocasa or dexmimicgen sim-eval in this MR except for a readme file.

Can you give me some details about how to deploy gr00t in dexmimicgen and remake the success rate in yours paper?

thanks a lot!