NVIDIA/Isaac-GR00T #143: Maniskill3 integration?

destin-v, 2025-04-28:
### 🚀 The feature, motivation and pitch

Are there any plans to integrate GR00T with [Maniskill3](https://www.maniskill.ai/).  I would like to use the premade simulation environments that Maniskill offers in combination with GR00T to test performance across different tasks.  

Maniskill has gymnasium APIs which follow the state-action pattern.  It also offers visual input from video feeds needed by GR00T.  Overall, I think it would be a valuable benchmarking, testing, and evaluation tool.

youliangtan, 2025-04-29:
There are no immediate plans for that, since the main sim tools we use are robocasa (mujoco based) and Isaac sim.

closed 2025-04-29 by youliangtan

destin-v, 2025-04-29:
Do you have examples of training pipelines leveraging Robocasa with different robotic types?  The documentation seems pretty sparse and it would be helpful to get a few more examples.

youliangtan, 2025-04-29:
@destin-v we are working on it. releasing soon