NVIDIA/Isaac-GR00T #145: Trying different chunk sizes​

alaanagyi, 2025-04-29:
why the inference time with chunk horizon 16 takes much time than chunk horizon 32, while it should be reversed??
[image]

[image]

[image]

alaanagyi, 2025-04-29:
the last one for chunk horizon 64, which make sense to  take longer time at each step, 
is it because:  The performance characteristics we see (88ms for horizon=32 vs 146ms for horizon=16) make sense because:​

When predicting with horizon=32, the model makes one forward pass to generate 32 future actions​

 With horizon=16, the model needs to make two separate forward passes to cover the same number of steps​

 So even though horizon=32 predicts more future steps at once, it's more efficient since it requires fewer total model invocations to cover the same trajectory length compared to horizon=16. This explains why the longer horizon actually runs faster in practice.​ but action horizon 64 says the opposite? can you illustrate why ?

youliangtan, 2025-05-11:
@alaanagyi  I cannot tell which time is which from your screenshots

closed 2025-07-17 by youliangtan