NVIDIA/Isaac-GR00T #152: Questions about history horizon and prediction horizon

robot-gen, 2025-05-06:
I would like to double-check how to achieve observation history horizon = 2 and the prediction action chunk is 16.

Should I just set the delta_indices of video_modality, state_modality, and action_modality such as:

```
video_modality = ModalityConfig(
      delta_indices=[-1,0],
      modality_keys=self.video_keys,
)
state_modality = ModalityConfig(
      delta_indices=[-1,0],
      modality_keys=self.state_keys,
)
action_modality = ModalityConfig(
      delta_indices=[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15],
      modality_keys=self.action_keys,
)
```
Anywhere else, should I also make changes? Thanks!

robot-gen, 2025-05-06:
I ran into the assertion "assert images.shape[0] == 1, "double check formatting when doing multi-time step" under ````_apply_gr00t_processing``. Has the code already supported multi-time step images processing, or I have to manually change this?

Yuheng2000, 2025-05-07:
same question

youliangtan, 2025-05-11:
I think this might not be fully supported as we usually just use single history. Feel free to open a PR to fix this, as this would be helpful.