NVIDIA/Isaac-GR00T #153: torch.OutOfMemoryError

Cupid0i0, 2025-05-06:
ENV: RTX4090 24G

cmd:
(base) root@4c4d9d6f3d39:/Isaac-GR00T# python scripts/gr00t_finetune.py    --dataset-path /gr00t_dataset/so100_2test/    --num-gpus 1    --output-dir ~/so100-checkpoints     --max-steps 1000    --data-config so100    --video-backend torchvision_av    --batch_size 1
/root/miniconda3/lib/python3.10/site-packages/albumentations/check_version.py:49: UserWarning: Error fetching version info The read operation timed out
  data = fetch_version_info()
2025-05-06 15:59:51.634182: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-05-06 15:59:51.634450: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-05-06 15:59:51.636372: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-05-06 15:59:51.644746: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-05-06 15:59:53.247936: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT

OutOfMemoryError:
The input hidden states seems to be silently casted in float32, this might be related to the fact you have upcasted embedding or layer norm layers in float32. We will cast back the input in torch.bfloat16.
Starting from v4.46, the `logits` model output will have the same type as the model (except at train time, where it will always be FP32)
Traceback (most recent call last):
  File "/root/miniconda3/lib/python3.10/site-packages/torch/optim/adamw.py", line 152, in _init_group
    state["exp_avg_sq"] = torch.zeros_like(
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 288.00 MiB. GPU 0 has a total capacity of 23.64 GiB of which 53.69 MiB is free. Process 12363 has 23.58 GiB memory in use. Of the allocated memory 21.98 GiB is allocated by PyTorch, and 1.14 GiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)

BouajilaHamza, 2025-05-10:
I have encountred the same issue .
It is may be the finetujing script is full model finetuning ?
I do not know if there is lora implementation or  qlora for finetuning .
Have you tried something like that? 
Also have you tried loading the model uaing bitandbytes ? 
Loading the model in lower precision may reduce the memory allocation a lot

BouajilaHamza, 2025-05-13:
Activating lora with low rank equal 1 and batch size 4 solved the problem for me

xwjiang2010, 2025-09-07:
The repo does support LoRA finetuning. Please take a look at the tutorial.
By default, full finetuning is used. If the error happens, you can try lower the batch size or lower the dataloader_num_workers. 

This is a CPU memory profile of a full finetuning run I did out of box. It climbs up during the beginning and pretty much stabilizes. 

[image]

Waircle, 2026-02-05:
It seems that Gr00t 1.6 doesn't come with built-in support for LoRA.