NVIDIA/Isaac-GR00T #154: Out of memory error when fine-tuning on runpod with 3 camera perspectives

robindehullu-howest, 2025-05-06:
### 🐛 Describe the bug

I am encountering an out of memory issue when fine-tuning on datasets with 3 camera perspectives. I am using runpod.io as cloud compute.

This is the command I use:
```bash
python scripts/gr00t_finetune.py \
	--dataset-path /workspace/data/so100/complete_dataset/ \
	--output-dir /mnt/gr00t_finetune_01  \
	--max-steps 500 \
	--data-config so100 \
	--video-backend torchvision_av \
	--dataloader_num_workers 4 \
```

This was tested on runpod.io using the following pod configurations with varying amounts of dataloaders:
- 	GPU:	A100 SXM (80GB VRAM)
	vCPUs:	10
	RAM:	125GB
-	GPU:	L40		(48GB VRAM)
	vCPUs:	8
	RAM:	94GB

Both systems result in an OOM error (Not CUDA) when decoding video inputs. This doesn't happen instantly when the amount of dataloaders is reduced. The "best" result, using a single DL, reaches ~250 steps before terminating.
There is plenty of memory available during the process and I don't believe it actually runs out of it.

The only alterations to the original image, available in this repository, were in the data_config where I edited the video keys to fit my dataset and modality.json:
```python
class So100DataConfig(BaseDataConfig):
    video_keys = ["video.top", "video.frontal", "video.gripper"] <----
    state_keys = ["state.single_arm", "state.gripper"]
    action_keys = ["action.single_arm", "action.gripper"]
    language_keys = ["annotation.human.task_description"]
    observation_indices = [0]
    action_indices = list(range(16))
```

I have also tested fine-tuning on a publicly available huggingface-hub dataset that uses 2 camera perspectives. The fine-tuning works fine out-of-the-box but when I duplicate a perspective to simulate a triple-perspective setup I encounter the same OOM error. My dataset has been recorded using the LeRobot framework.

### Versions

```bash
Python 3.11.11
absl-py==2.2.2
accelerate==1.2.1
aiosignal==1.3.2
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
archspec @ file:///home/conda/feedstock_root/build_artifacts/archspec_1737352602016/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1733250440834/work
astunparse==1.6.3
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1737819173731/work
av==12.3.0
beautifulsoup4 @ file:///home/conda/feedstock_root/build_artifacts/beautifulsoup4_1733230845337/work
blessed==1.21.0
blessings==1.7
boltons @ file:///home/conda/feedstock_root/build_artifacts/boltons_1733827268945/work
Brotli @ file:///home/conda/feedstock_root/build_artifacts/brotli-split_1725267488082/work
cachetools==5.5.2
certifi @ file:///home/conda/feedstock_root/build_artifacts/certifi_1734380492396/work/certifi
cffi @ file:///home/conda/feedstock_root/build_artifacts/cffi_1725560564262/work
chardet @ file:///home/conda/feedstock_root/build_artifacts/chardet_1724954797915/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1735929714516/work
click @ file:///home/conda/feedstock_root/build_artifacts/click_1734858813237/work
cloudpickle==3.1.1
cmake==3.31.4
colorama @ file:///home/conda/feedstock_root/build_artifacts/colorama_1733218098505/work
conda @ file:///home/conda/feedstock_root/build_artifacts/conda_1737546925511/work
[... 185 lines cut ...]
torch==2.5.1
torchaudio==2.5.1
torchelastic==0.2.2
torchvision==0.20.1
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1735661334605/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
transformers==4.45.2
triton==3.1.0
truststore @ file:///home/conda/feedstock_root/build_artifacts/truststore_1729762363021/work
typeguard==4.4.2
types-dataclasses==0.6.6
typing-inspection==0.4.0
typing_extensions==4.13.2
tyro==0.9.17
tzdata==2025.2
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1734859416348/work
wandb==0.18.0
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1732827521216/work
zmq==0.0.0
zstandard==0.23.0
```

robindehullu-howest, 2025-05-07:
Update:
I tested smaller batch sizes. Until now I left it on default (16).
The batch size has the same overall effect as the number of dataloaders. Reducing it just means delaying the inevitable.

I am guessing this means I will encounter it as well with 1 or 2 camera perspectives but I haven't trained it for long enough yet.

agvdndor, 2025-05-08:
Linked to #119

youliangtan, 2025-05-11:
What OOM are you getting? is it GPU or CPU. You can paste your logs here. 

Not sure if this info helps, but if I recall I did encounter some OOM issue with videos before, especially when we do caching for perf. One thing we also resize the lerobot dataset videos to a smaller size to maximize performance and minimize mem usage.

robindehullu-howest, 2025-05-12:
This is the full call stack including the OOM.
My current workaround involves transcoding the video files from av1 to h264 in order to use the decord video backend. 

```bash
  2%|█▎                                                               | 40/2000 [00:46<34:48,  1.07s/it]Traceback (most recent call last):
  File "/workspace/scripts/gr00t_finetune.py", line 232, in <module>
    main(config)
  File "/workspace/scripts/gr00t_finetune.py", line 204, in main
    experiment.train()
  File "/workspace/gr00t/experiment/runner.py", line 164, in train
    self.trainer.train(resume_from_checkpoint=self.resume_from_checkpoint)
  File "/workspace/gr00t/experiment/trainer.py", line 153, in train
    return super().train(resume_from_checkpoint, trial, ignore_keys_for_eval, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2052, in train
    return inner_training_loop(
           ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/transformers/trainer.py", line 2345, in _inner_training_loop
    for step, inputs in enumerate(epoch_iterator):
  File "/opt/conda/lib/python3.11/site-packages/accelerate/data_loader.py", line 563, in __iter__       
    next_batch = next(dataloader_iter)
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 701, in __next__  
    data = self._next_data()
           ^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 1465, in _next_data
    return self._process_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 1491, in _process_data
    data.reraise()
  File "/opt/conda/lib/python3.11/site-packages/torch/_utils.py", line 714, in reraise
    raise RuntimeError(msg) from None
RuntimeError: Caught MemoryError in DataLoader worker process 1.
Original Traceback (most recent call last):
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/_utils/worker.py", line 351, in _worker_loop
    data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
           ^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch    
    data = [self.dataset[idx] for idx in possibly_batched_index]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
            ~~~~~~~~~~~~^^^^^
  File "/workspace/gr00t/data/dataset.py", line 488, in __getitem__
    return self.transforms(self.get_step_data(trajectory_id, base_index))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/data/dataset.py", line 522, in get_step_data
    data[key] = self.get_data_by_modality(trajectory_id, modality, key, base_index)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/data/dataset.py", line 782, in get_data_by_modality
    return self.get_video(trajectory_id, key, base_index)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/data/dataset.py", line 652, in get_video
    return get_frames_by_timestamps(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/gr00t/utils/video.py", line 112, in get_frames_by_timestamps
    for frame in reader:
  File "/opt/conda/lib/python3.11/site-packages/torchvision/io/video_reader.py", line 200, in __next__  
    frame = next(self._c)
            ^^^^^^^^^^^^^
  File "av/container/input.pyx", line 208, in decode
  File "av/packet.pyx", line 80, in av.packet.Packet.decode
  File "av/video/stream.pyx", line 41, in av.video.stream.VideoStream.decode
  File "av/video/stream.pyx", line 50, in av.video.stream.VideoStream.decode
  File "av/codec/context.pyx", line 462, in av.codec.context.CodecContext.decode
  File "av/codec/context.pyx", line 238, in av.codec.context.CodecContext.open
  File "av/error.pyx", line 326, in av.error.err_check
av.error.MemoryError: [Errno 12] Cannot allocate memory
```

a1037441813, 2025-05-19:
you can  set  num_worker =1  ,,,,,,,,,,,,,,,,,,,,,,,but training speed will be slow

robindehullu-howest, 2025-05-26:
I have tested setting `num_worker = 1`

The result, for me, is basically the same. It takes a lot longer to train but it still experiences the OOM error.
The added delay just results in additional compute cost while making little difference in the total steps reached overall.

I have however tried transcoding the video files to allow for decord as the video backend, as suggested in #119. 
> I use decord as video backend seem solve the problem

I tried to avoid having duplicate data stored. For that reason I transcoded the videos on the pod after having pulled the dataset from cloud storage. This didn't result in any issues until I implemented multi-threaded transcoding to speed up the process. While monitoring the memory usage I can see that it peaks at 6% before resulting in the same error experienced during fine-tuning.

The transcoding script is totally isolated from the GR00T-N1 repository. This leads me to believe that the issue does not stem from GR00T-N1 or AV1 but lies with cgroup limits on runpod.

My current solution for prerecorded data is to transcode the videos locally. That does result in the duplicate data but oh well... From now on I will be using h264 as video encoder when recording datasets for decord compatibility.

closed 2025-05-26 by robindehullu-howest