NVIDIA/Isaac-GR00T #155: Guidance on Integrating Isaac GR00T with Custom URDF-Based Humanoid Robot

BouajilaHamza, 2025-05-07:
I'm working on integrating NVIDIA Isaac GR00T with a custom URDF-based humanoid robot within PyBullet from company. While I've successfully imported the URDF into Pybullet and manipulalted the urdf using Isaac Gr00t n1 i am seeking guidance on the following :

Joint Configuration Alignment: Ensuring that GR00T's output commands are compatible with my robot's joint configurations and limits.

Simulation Optimization: Recommendations for optimizing simulation performance and accuracy when using GR00T with custom robot models.

Any detailed guidance, documentation, or examples related to these points would be greatly appreciated.

youliangtan, 2025-05-11:
This is not so relevant to Gr00t N1, the URDF is simulation related and the model will model the distribution according to your data. (e.g. we directly use min-max normalization for abs joint control). You can also customize the normalization scheme according to the respective joint limits. 

On top of that, we are releasing the sim soon. so could refer to that soon.

closed 2025-05-11 by youliangtan

BouajilaHamza, 2025-05-11:
Okey thank you