NVIDIA/Isaac-GR00T #156: Please share info about the calibration of SO100 for dataset so100_strawberry_grape

jpizarrom, 2025-05-07:
### 📚 The doc issue

@youliangtan can you please share info about the state representation for SO100 for the dataset so100_strawberry_grape, what value types were used for observation.state.

e.g. in your dataset the values of shoulder_pan are centered on `90` [sample](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2Fyouliangtan%2Fso100_strawberry_grape%2Fepisode_0%3Ft%3D6), but for other SO100 dataset are centered on  `0` for similar position [sample](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2Fcadene%2Fso100_test_5%2Fepisode_1%3Ft%3D2)

there is any particular reason for that state range representation?

Thanks

youliangtan, 2025-05-11:
This is the calibration file for reference.
[main_follower.json](https://github.com/user-attachments/files/20144380/main_follower.json)

According to the provided `meta/modality_config.json`, we use all states of 6 DOF of the arm as `observation.state`

I'm not sure about the 90degree switch in so100, however the same dataset is able to replay on my setup. This is the sanity check which users should try. Also, the example meant to serve as a fine-tuning example for users to tryout the fine-tuning. Ideally users should always collect your own trajectories for fine-tuning to minimize the distribution gap between each robot setup.

closed 2025-05-11 by youliangtan

jpizarrom, 2025-05-11:
@youliangtan Thanks for the calibration files. I am trying to use your dataset as sanity check with a real so100, because with my own datasets I am not getting good results, maybe the reason is that i am using lora, and lora is being applied to all of the model, and not only the the modules defined by the parameters `tune_XXX`.

youliangtan, 2025-05-12:
Yes LoRA performs much poorly from my limited experiments.