NVIDIA/Isaac-GR00T #157: Issues in fine-tuning and deploying Groot on 7 DoF franka arm

vineet2104, 2025-05-08:
Hi authors,

Thank you so much for open-sourcing your work!

I have tried fine-tuning Groot for a 7 DoF franka arm using a small manually collected dataset. However I am facing some issues in replicating the performance as shown on the smaller SO-100 arm. 

My dataset comprises 4 tasks, each with 30 demonstrations. On average every demonstration has around 80-90 steps. These tasks involve simple pick and place language instructions. I recorded the egocentric camera view, hand camera view, robot joint angles (for states) and velocity commands (for actions). I fine-tuned the Groot-N1 base model with default settings after converting my dataset into the LeRobot format. My action vector is 8D, with the last dimension being the gripper state (1/0).

I have attached the visualization of the inference results on one of the tasks seen by the model during fine-tuning (45K steps). The predictions are very noisy and although they follow somewhat a general pattern of the ground truth actions, they are not deployable on the real robot. There are some joints that remain constant throughout the task, and the policy is not able to learn that.

I was wondering if this behaviour was observed by someone else. Does the step-function like behaviour of my ground truth action vector cause any learning issues? 

I'd be grateful for any insights or suggestions to solve this problem. 

Thank you!

[image]

closed 2025-05-09 by vineet2104

eladb, 2025-05-09:
Curious if you can share your conciliations.

vineet2104, 2025-05-09:
Hey, sure. I made a mistake during computing the norm of the state and action vectors for my custom dataset. I’d recommend ensuring your data files structure looks exactly the same as the SO100 dataset provided, and then let the data loader handle building stats.json file while fine-tuning for the first time.