NVIDIA/Isaac-GR00T #160: pip install -e . failed due to missing pyav distribution on Ubuntu 22.04 (Python 3.10)

peterwang779, 2025-05-09:
I followed the instructions to set up the GR00T environment using:

- OS: Ubuntu  22.04

- Python: 3.10 (via Conda)

- CUDA: 12.4

Installation command:

1. conda create -n gr00t python=3.10  

2. conda activate gr00t  

3. pip install --upgrade setuptools  

4. pip install -e .

The installation failed with the following error:

**ERROR: Could not find a version that satisfies the requirement pyav (from gr00t) (from versions: none)  
ERROR: No matching distribution found for pyav**

Below is the full installation log from my [setup:]

[log.txt](https://github.com/user-attachments/files/20118477/log.txt)

BouajilaHamza, 2025-05-09:
It is more likely the package was removed from pypi .
The issue is new .
It worked for me by just removing it from dependencies .
There is a package named av in dependencies has the same role as pyav i think .

BouajilaHamza, 2025-05-09:
Also i do recommend to use uv 
````
pip install uv
uv venv #this will create .venv folder 
source .venv/bin/activate
uv remove pyav 
````
If an error occured from tensorflow about python version .
Add a file in the root folder named .python-version
That contains 
`3.10`
Also there are some dev dependencies if you are gonna test or develope .
Use 
```
uv sync --all-extras
```

closed 2025-05-13 by peterwang779